barbeau/gpstest #408: Persian localization

iriman, 2020-05-25:
**Summary:** 

Persian localization is done via Transifex. I would be grateful if you can update it in the app.

barbeau, 2020-05-26:
@iriman Thanks for working on this! I'll try to get it into the next update. I haven't had as much time to work on the app recently, but hopefully soon 🤞 .

barbeau, 2020-06-03:
@iriman Thanks for doing the Persian translation! I just added the code and I'll try to get out a new beta soon with this. Please sign up as a beta tester if you haven't yet so you can take a look at it in the app:
https://github.com/barbeau/gpstest/blob/master/BETA_TESTING.md#register-as-a-beta-tester 

Two quick questions:
* I used `زبان فارسی` for the Persian language in the list of available languages in the Settings. Is this right?
* How would you liked to be listed in the Acknowledgements section? I currently have you listed as `Iman`.

closed 2020-06-03 by barbeau

iriman, 2020-06-03:
Thank you @barbeau. I opt-in to the beta test and waiting for Play Store to make it available to me.

About questions:
1.  That's correct. However, `فارسی` is enough, `زبان` means `language`.
2. Good. I prefer `Iman (imni)`