barbeau/gpstest #409: Add GPS date field to the display

m31d4ri0n, 2020-05-29:
**Request:**
When using the application we would benefit from being able to see the GPS date, time and location data fields on the screen at the same time.

**The Why:**
The use of your application would help combat illegal poaching as part of anti-poaching operations aboard a fleet of vessels.

The current requirements for making photographic evidence requires that a photographer takes a photo of a handheld GPS unit displaying the following data fields from GPS:

1. Date
2. Time
3. Co-ordinates (50°32.3’N 03°25.4’W)

These are visible on the device in the picture, whilst the illegal vessel is in the background of the image. This photographic evidence is then perfect for use in court.

Presently a Garmin 72H is being used but they are now discontinued due to smartphones becoming so popular, meaning we have issues.

E.G.
[Fig. 0 - Example of Photo (vessel in background not normally blurry)]
https://www.dropbox.com/s/lq4dnxj9tnpew11/Screenshot%202020-05-29%2018.21.50.png?dl=0

**How This Could Be Fullfilled:**

The addition of DATE in the Status page of the application
[Fig. 1 - GPS App]
https://www.dropbox.com/s/zlgp21s1fv0lplt/Screenshot%202020-05-29%2017.57.23.png?dl=0

**Further Thoughts** 
Taking photos at sea, sometimes in low light conditions requires slower shutter speeds on cameras. Movement on the vessels, plus the slow shutter speed leads to blur on an image. 

My concern is with the current aesthetics of your app, with DATE added to the status page, probably the text will still be too small for the camera to display clearly.

I don't want this work to be for nothing so I am trying to be as thoughtful as possible.

** How this could be mitigated:**

Could be possible to have a customisable page where user could select what data to display from GPS. 

This, I completely appreciate, is a much bigger ask.

With smartphones pushing against the equipment we use we're in an awkward position moving forward without any appropriate apps.

barbeau, 2020-05-29:
Thanks @m31d4ri0n for the feedback, and I appreciate the context of the use case! Let me think about the best way to accomplish this. The larger font concept is also related to https://github.com/barbeau/gpstest/issues/176.

Would adding the date and time to the "Share" screen (tap the arrow in the top blue action bar after getting a fix) below the existing lat/long work?

I could add a "Include date/time" checkbox under "Include altitude" that when checked would show the date/time under the lat/long in same font size as the lat/long.

[image]

m31d4ri0n, 2020-06-01:
_Would adding the date and time to the "Share" screen (tap the arrow in the top blue action bar after getting a fix) below the existing lat/long work?_

Due to the nature of initial arrests of illegal vessels, it's a disadvantage to have any photographic evidence which can be disputed and there are **big** **challenges** in getting criminals into a court of law to face scruitiny.

With that in mind we are looking for something which, on a photograph, would need no explination and would be very clear that the law has been broken; in one picture the entire story is told. 
[image]

Out at sea we do not have the luxury to dive into files or metadata and what is pictured must be something that can be observed as valid / official.

I hope that answers your question and I hope the feedback is of some value.

barbeau, 2020-06-01:
@m31d4ri0n Here's an example screenshot of what date would look like as part of the Share screen:

[image]

What I'm hearing you say is that for your use case this screen is too busy and not clear enough. Is that correct?

m31d4ri0n, 2020-06-02:
Hi Barbeau,

_What I'm hearing you say is that for your use case this screen is too busy and not clear enough. Is that correct?_

The share screen loses a lot of it's 'official' feel. It no longer looks like an official tool. It looks like a share screen which could be debated.

The advantage you have with the status screen is it shows active satellites and is part of your application, running live. Whilst it's not ideal, it looks a lot better than the Share screen.

I'm sorry if I'm not able to convey what I mean by this. But when you are trying to get authorisation to arrest an illegal vessel then you need the evidence to be clear, undebateable and 100% believable in the moment and people need to believe they are completely doomed by an official looking bit of technology. 

The garmin has that. At least it did. Until they stopped making them and lost interest because of smartphones haha.

barbeau, 2020-06-02:
> The share screen loses a lot of it's 'official' feel. It no longer looks like an official tool. It looks like a share screen which could be debated.
>
> The advantage you have with the status screen is [...]

@m31d4ri0n Thanks, that definitely makes sense.

My original thought (prior to considering the low light and motion environment) was to simply add the date next to the time, so instead of:

`Time: 17:57:16`

...it would look something like:

`Time: 17:57:16, Jul 4, 2020`

I'll likely only add the date on screens that are wide enough to accommodate it in portrait.

With this design, to help with the low light and motion you could increase the Android system font size:
https://support.google.com/accessibility/android/answer/6006972?hl=en

If the data spills off the side of the screen with a larger font size in portrait, you could rotate the phone to landscape and get more room.

Do you think the above would work for your use case?

You could try increasing the system font size with the existing app Status screen to get an idea of how this would look.

goldfndr, 2020-06-11:
Displaying "EST" instead of "EDT" during DST would appear to be a bug (that might be adequate for a dispute), unless the device is in a location within the Eastern Timezone that doesn't observe DST.

barbeau, 2020-06-11:
@goldfndr if you're referring to my example screenshot above, you can ignore that - I just typed in the text to fill the mockup, and accidentally wrote "EST".

barbeau, 2020-06-16:
@m31d4ri0n A pull request for this feature with screenshots is available at https://github.com/barbeau/gpstest/pull/412, if you want to check out what it looks like on a Samsung Galaxy S8+.

I'd like to do a little more testing on other screen widths to see if I've gotten the 450dp threshold about right.

closed 2020-06-17 by barbeau

barbeau, 2020-06-23:
@m31d4ri0n This feature is live in the v3.7.0 beta if you want to check it out - https://groups.google.com/forum/#!topic/gpstest_android/gaI2NJiDhFo.

PR #412 (2020-06-17): feat: Add date to Status view for wide screens
This adds date to the Status view when the screen width is more than 450dp (including when the screen is rotated to landscape).

Closes #409.

Here are example screenshots from a Samsung Galaxy S8+ - the date doesn't appear when in portrait but appears in landscape:

[image]

[image]

TODO:
- [x] Test on wider screens to see if threshold is about right
commit eed61068 (2020-06-17): feat: Add date to Status view for wide screens (#412)

This adds date to the Status view when the screen width is more than 450dp (including when the screen is rotated to landscape).

Closes #409.