barbeau/gpstest #410: Conh1976

CongPhungHuynh, 2020-06-03:
`**Summary:**` 

Summarize your issue in one sentence (what goes wrong, what did you expect to happen)

**Steps to reproduce:** 

How can we reproduce the issue?

**Expected behavior:** 

What did you expect the app to do?

**Observed behavior:** 

What did you see instead?  Describe your issue in detail here.

**Device and Android version:** 

What make and model device (e.g., Samsung Galaxy S3) did you encounter this on?  What Android version (e.g., Android 4.0 Ice Cream Sandwich) are you running?  Is it the stock version from the manufacturer or a custom ROM?  What version of Google Play Services is running on your device?

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.

barbeau, 2020-06-03:
Closing this out for cleanup - @CongPhungHuynh let me know if opening this issue wasn't accidental, there isn't any content.

closed 2020-06-03 by barbeau

CongPhungHuynh, 2020-06-28:
cong Phung HUYNH