barbeau/gpstest #413: Update Russian translation

shooreek, 2020-06-23:
**Describe the bug**
In Russian translation one link is broken, wording could be more accurate

**To Reproduce**
Steps to reproduce the behavior:
1. Get device vulnerable to GPS week number rollover issue
2. Tap on the highlighted and incorrect time to see message explaining what's wrong

**Expected behavior**
Everything is fine.

**Observed behavior**
- Link is followed by a dot which is considered as part of the link, clicking on such broken link won't open any page
- Wording is not so accurate, instead of "переключение" (switch) I would use "сброс" (reset)

**Screenshots:** 
[image]

Here is the fixed version of strings.xml file:
[strings.zip](https://github.com/barbeau/gpstest/files/4816210/strings.zip)

P.S. www.transifex.com site is not in sync with the master branch, I had to download the file from GitHub. Is that normal? It is not reflected in the translation guide.

barbeau, 2020-06-23:
@shooreek Thanks for the report! Hmmm...you should be able to make these changes on Transifex, and then I can pull the strings into the project from there. Are you saying the above string isn't the same at https://www.transifex.com/sean-barbeau/gpstest-android/language/ru/?

shooreek, 2020-06-23:
@barbeau, I did my changes to this string on Transflex and grabbed the file via "Download for use". But it turned out that grabbed file has few other differences except for my change: `main_help_whatsnew` and `help_text` strings were missing in Transflex file but present in master. 

So I took file from master and updated it so one can simply take it and commit. If you prefer to pull from Transflex you can do that. Don't know what do you mean by saying "pull" but if you're going to take entire file you may accidentally remove `main_help_whatsnew` and `help_text` strings that are already in master.

barbeau, 2020-06-23:
@shooreek Thanks for the info.

Yes, when I update English strings on GitHub (e.g., whats new and help), Transifex automatically removes them from translations in other languages so they can be re-translated. When I run the command-line tool to pull changes from Transifex into my local GPSTest project, I get all the new translations, but then I usually keep some of the old translations for things like help that are massive until they get updated in Transifex.

So I'll go ahead and pull your changes from Transifex (and ignore stuff that's missing) for the next beta release - I'll leave this issue open to remind me. Thanks again!

ValZapod, 2020-06-28:
It is now correctly translated https://www.transifex.com/sean-barbeau/gpstest-android/translate/#ru/strings-xml/194446402?q=text%3Agps-week so next release will have correct text.

commit 75987160 (2020-07-17): fix: Update Russian translation

Closes #413
closed 2020-07-17 by barbeau