barbeau/gpstest #420: Update app to target Android 10 (API 29)

barbeau, 2020-07-23:
**Is your feature request related to a problem? Please describe.**
When publishing update to Google Play I now see the message:

>Your app currently targets API level 28 and must target at least API level 29 to ensure it is built on the latest APIs optimized for security and performance.
>
>From August 2020, new apps must target at least Android 10 (API level 29).
From November 2020, app updates must target Android 10 (API level 29).
>
>Tip: Change your app's target API level. Learn how. https://developer.android.com/distribute/best-practices/develop/target-sdk.html

**Describe the solution you'd like**
Update to target API 29

**Describe alternatives you've considered**
None

paulle, 2020-08-18:
Why not update to android11, API level 30, because android11 will be released in some weeks. This would avoid problems for users getting the update fast or getting a new phone running android11 and there would be no need updating the API level for a year or so.

barbeau, 2020-08-18:
@paulle So "targeting Android 10" and "supporting Android 10" are two different things in the Android world. GPSTest currently supports Android 10 and 11, and should run on both of them (and future Android versions) without problems. 

Increasing the `targetSdkVersion` in the app, which is what this issue refers to, means that the developer has made changes to the app required by Google to modernize it in some ways (if the app uses the features that Google has changed). GPSTest is fairly minimal in scope in terms of using platform APIs so there may not even be any code changes to accomplish this. If you're running GPSTest on a version of Android ahead of the current `targetSdkVersion` and a feature that GPSTest uses was changed, Android will run the app in a type of compatibility mode. This usually isn't visible to the user so you wouldn't know the difference.

I'll certainly look at targeting Android 11 as soon as Google releases guidance on this, but Android 11 isn't even listed on the compatibility page yet:
https://developer.android.com/distribute/best-practices/develop/target-sdk.html#pre10

paulle, 2020-08-18:
Thank you for explanation.

-------- Original-Nachricht --------

PR #432 (2020-09-01): chore: Bump compile/targetSdkVerion, build tools
Also changes to canonical GnssStatus.CONSTELLATION_IRNSS, and therefore closes #340.

Closes #340
Closes #420
commit f0753380 (2020-09-01): chore: Bump compile/targetSdkVerion, build tools (#432)

* Also changes to canonical GnssStatus.CONSTELLATION_IRNSS, and therefore closes #340.
* Fix tests by handling nullable bundle extras

Closes #340
Closes #420
closed 2020-09-01 by barbeau