barbeau/gpstest #421: Could you show if the carrier-phase is valid in your app？

Nronaldo, 2020-07-28:
**Is your feature request related to a problem? Please describe.**
I find some smartphones confirm they support dual-frequency GNSS and shows dual-frequency GNSS in you app. However, we cannot read the the carrier-phase with "getAccumulatedDeltaRangeMeters" API.   I test Xiaomi Mi 9 Xiaomi Mi10 OnePlus 7. They cannot output carrier-phase. 
I wonder which smartphones can provide valid carrier-phase for RTK except for Xiaomi Mi 8. Could you please give me some advice?

barbeau, 2020-08-06:
@Nronaldo Thanks for the feedback! I agree, it's hard to tell this right now, especially since it appears Google has removed the supported devices list from https://developer.android.com/guide/topics/sensors/gnss#supported_devices (I *think* `getAccumulatedDeltaRangeMeters()` support was listed there?).

I think showing this information in GPSTest would work well within the new Device Info tab I've been working on - I've added it to the possible list of features at https://github.com/barbeau/gpstest/issues/313#issuecomment-513276069.

barbeau, 2021-06-03:
Closing this as a duplicate as it will be part of https://github.com/barbeau/gpstest/issues/313

closed 2021-06-03 by barbeau