barbeau/gpstest #422: Support for IOS

tucotuco, 2020-07-28:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

We use GPS Test extensively when teaching courses on georeferencing, and have even made a video to support these activities. In some countries iPhones are used with enough frequency that some of our sutedents are left without a good alternative that we know of. When teaching in person, this isn't so much of an issue because the students can share devices, but in the current situation where we have to do the courses online, it leaves those with iPhones in a bind. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

A version that functions on iPhones. ;-)

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Other applications with the same important features that GPS Test has, i.e., choose coordinate reference system, choose coordinate format, choose vertical datum (geoid or ellipsoid), shows theoretical accuracy.

**Additional context**
Add any other context or screenshots about the feature request here.
https://www.youtube.com/watch?v=96tIRsuGPXc

barbeau, 2020-08-05:
@tucotuco Thanks for reaching out! Glad it's helpful, I actually started working on this app when teaching a GPS class myself. 

The only open-source iOS GPS apps I know of are here - https://github.com/barbeau/awesome-gnss#ios-apps-open-source.

Unfortunately my iOS development skills are basically non-existent. I'd love to work on an iOS version of GPSTest, especially if we could figure out a way to cross-compile the currently code to iOS (e.g., by doing a complete Kotlin conversion?). My bandwidth is pretty limited, though, as this is entirely a side-project for me, so that's my main constraint.

Let me know if you (or anyone else) find a good iOS equivalent. Any additional thoughts are welcome.

One note after looking at the video - note that this repository is for "[GPSTest](https://play.google.com/store/apps/details?id=com.android.gpstest)" not "[GPS Test](https://play.google.com/store/apps/details?id=com.chartcross.gpstest&hl=en_US)" - yes, it's confusing. Just wanted to clarify that.

tucotuco, 2020-08-12:
@barbeau Thanks Sean for the response. After some searching we were able to find the iGeoTrans X Lite in the App Store (https://apps.apple.com/us/app/igeotrans-x-lite/id471361224). I'm not able to assess it because I don't have access to an iPhone (in quarantine in the mountains in Patagonia), but others in the course confirm that it has the characteristics we need for the course. It would indeed be great to have GPSTest for iPhone as well, not only for consistency, but also because I think GPSTest is a very good product.

barbeau, 2020-09-14:
Kotlin Multiplatform Mobile is now in Alpha:
https://blog.jetbrains.com/kotlin/2020/08/kotlin-multiplatform-mobile-goes-alpha/

barbeau, 2020-11-05:
Netflix example using KMM:
https://netflixtechblog.com/netflix-android-and-ios-studio-apps-kotlin-multiplatform-d6d4d8d25d23

gdt, 2020-11-30:
I know someone who is geo-clueful and has an iPhone, and they have been unable to get any information beyond the location API (which I think is lat/lon/elevation/accuracy/speed/course sorts of things).   I am aware of no reason to believe that information about satellites being tracked is available in any way.

Until there is a pointer to Apple docs that provide an API to get the information that a program like GPSTest or SatStat needs, I recommend closing this bug.   I'd encourage the submitter to file an issue with Apple "Location API does not provide per-satellite tracking information" :-)

barbeau, 2020-12-01:
@gdt Good point. That reflects the most recent state of my knowledge on iOS as well, but it's been a while since I looked into it.

So for anyone who is interested in this, please file an issue with Apple requesting an API for per-satellite data. And the first person that does this, please add a link in a comment here so others can upvote.

barbeau, 2021-09-27:
Someone shared the iOS "GPS Plan" app with me:
https://apps.apple.com/us/app/gps-plan/id623849032

I assume this is fetching almanac and ephemeris data from other sources because AFAIK iOS still doesn't support the on-device APIs to get a lot of this data directly from the device hardware, including what satellites are in use.

I'd be curious to get feedback from anyone who has used it to better understand what the app is actually showing.

closed 2022-11-02 by stale[bot]

barbeau, 2024-05-01:
Here is a link to the iOS developer info showing raw measurements still aren't available - https://developer.apple.com/documentation/corelocation/cllocationmanagerdelegate/1423615-locationmanager