barbeau/gpstest #424: Can't edit Accuracy view longitude and latitude

gnusuari0, 2020-08-05:
**Describe the bug**
Accuracy view 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to accuracy view.
2. Tap on edit.
3. Tap on longitude or latitude textview.
4. Nothing happens.

**Expected behavior**
Software keyboard should appear, allowing to type coordinates.

**Observed behavior**
A clear and concise description of what actually happened.

**App, Device and Android version:** 
v3.7.2 (18069-osmdroid), OnePlus 7 Pro, Android 10, stock ROM Oxygen OS 10.0.6.GM21BA. Goole Services Framework version 10-5771379.

**Screenshots:** 

[image]

gnusuari0, 2020-08-05:
And as you can see, OSM map does not load. The Map view works as expected.

barbeau, 2020-08-05:
@gnusuari0 Thanks for reporting this! I noticed that something strange happened too - I updated the Android library used for the top card view it appears something broke in the process. I'll take a look.

barbeau, 2020-08-05:
Looks like this was a bug in the `constraint-layout:2.0.0-beta6` library release.

Changing to `implementation 'androidx.constraintlayout:constraintlayout:2.0.0-rc1'` fixes this.

closed 2020-08-05 by barbeau

barbeau, 2020-08-07:
This is fixed in the latest v3.7.3 release which is now in beta, if you want to test it out!

gnusuari0, 2020-08-08:
Nice.I'm an F-Droid user... I'll wait for the app to be published there.

commit 0e56007e (2020-08-05): fix: Update to ConstraintLayout v2-rc1 so MotionLayout works

Closes #424