barbeau/gpstest #425: Log file has duplicate lines for "Fix" entries

YoavZangvil, 2020-08-06:
**Describe the bug**
When enabling the logging "Location" option, the created log file has doubled entries for "Fix" entries

**To Reproduce**
1. Enable log
2. select Location
3. Run the app for a few minutes 
4. Close the app
5. Open the log file in a text viewer

**Expected behavior**
Each "Fix" line should be unique.

**Observed behavior**
There are double lines each time for each "Fix" entry

**App, Device and Android version:** 

App version 3.6.4
Devices observed: Huawei P20, Xiaomi Mi8, Samsung S8,S9,S10, Honor X10, Oppo Reno 10, Vivo X27
**Screenshots:**

YoavZangvil, 2020-08-06:
Attached is a screenshot of the log file.

[image]

barbeau, 2020-08-06:
Thanks for reporting this @YoavZangvil! The fix will be in the next beta release. If you're not signed up for beta updates and would like to check it out, see https://play.google.com/apps/testing/com.android.gpstest.

closed 2020-08-06 by barbeau

barbeau, 2020-08-07:
This is fixed in the latest v3.7.3 release which is now in beta, if you want to test it out!

commit 1cfba339 (2020-08-06): fix: Remove duplicate Fix entries from file log

Fixes #425