barbeau/gpstest #426: Support for NTRIP Web source

mapdegree, 2020-08-09:
**Is your feature request related to a problem? Please describe.**
We need NTRIP support for improving our positional accuracy in real time (RTK). 

**Describe the solution you'd like**
An NTRIP endpoint with credentials authentication. 

**Describe alternatives you've considered**
Not really applicable, NTRIP is an open technology standard and very helpful in RTK

**Additional context**
I would be happy to pay a once off for such functionality

barbeau, 2020-08-28:
@mapdegree Thanks for the feedback! So right now GPSTest doesn't calculate the position itself from raw measurements, it just shows the position calculated by the device hardware. So this would require adding calculating the position from raw measurements to the app as well. Given this is a side project for me I don't know how soon that would happen, but I'd certainly love to collaborate with someone on it to speed things up.

barbeau, 2020-09-30:
Another user pointed to this SDK:
http://flamingosdk.com/

Avamander, 2020-10-20:
This sounds really nice, if at all possible.

gdt, 2020-11-23:
I think it's best if GPSTest does not cross over into computing RTK positions.  It's a diagnostic tool that displays a GUI, which is entirely appropriate.   If one wanted to do RTK, presumably that position would then be usable by things like OsmAnd and Vespucci, so you'd get into being a high-accuracy network provider or a faux GPS provider via mocking, or some other interface.   The RTK process should be able to run in the background.  This could be a port of rtklib with the NDK.

As for flamingo, it appears to be "coming soon" and there is not even a hint on the website that it will be open source.  I could not find a link to source code, for example.   So it does not seem useful in an open source context, at least until there is source code under a Free license.

codeandfix, 2022-06-27:
yes, this would be fine, also with external GPS support over bluetooth/usb.
maybe for garmin gpsmap devices.

closed 2023-08-02 by stale[bot]