barbeau/gpstest #430: [FR] Adaptive icon support

staticssleever668, 2020-08-27:
I want support for adaptive icons to be added as this is like an eyesore - in the app list where most of the icons are circle there are a few that doesn't fit the style. Most modern Android apps support it.

barbeau, 2020-08-27:
@staticssleever668 Thanks for the reminder! That's something I can work on. Out of curiosity, could you share a screenshot of how this currently looks on your device, with you device make, model, and Android version?

staticssleever668, 2020-08-27:
Sorry, but because of my stupid privacy reasons I can only give you these
[image]
[image]

JonnyTech, 2020-08-27:
@barbeau here is a selection from my Android 9 device
[image]

closed 2020-08-31 by barbeau

barbeau, 2020-09-01:
@JonnyTech @staticssleever668 Adaptive icon support is now in the beta release for v3.8.0 - could you please check it out and see if it looks better on your devices?

Sign up for betas from Google Play at:
https://play.google.com/apps/testing/com.android.gpstest

JonnyTech, 2020-09-01:
Thanks @barbeau, but sorry but I do not use Google Apps so cannot access the beta. Can you provide a prebuilt apk?

barbeau, 2020-09-01:
@JonnyTech Sure, see https://github.com/barbeau/gpstest/releases/download/v3.8.0/osmdroidRelease-v3.8.0.apk

Giolly90, 2020-09-01:
It's working fine for me!
[image]
[image]
[image]
[image]

JonnyTech, 2020-09-01:
Perfect here too - great job, thanks Sean!

[image]

barbeau, 2020-09-01:
Thank for testing - glad it works! 😅

commit 39f10f79 (2020-08-31): feat: Add adaptive icon support

Source files are in the /icons directory

Closes #430