barbeau/gpstest #442: Accuracy Enhancements

gb-dursley, 2020-10-12:
**Is your feature request related to a problem? Please describe.**
I'm particularly interested in Height Accuracy - but:-
 Status screen only shown E H Acc - could E V Acc be added please?
  Accuracy screen 
    scale for grid
    + or - for V error
  Your...(position pane)
    in EDIT mode, the SAVE button is at the edge of the pane, leading to accidental taps on the grid (changing position!) rather than the Save button. Suggest moving the button to the white space near the top right corner of the pane.

barbeau, 2020-10-13:
@gb-dursley Thanks for the feedback! Having multiple problems/enhancements bundled in one issue makes it hard to track them. Let's try to figure out which items need their own issues and then close this one out.

>Status screen only shown E H Acc - could E V Acc be added please?

What device are you using? Unfortunately not all devices support vertical accuracy info, but for the ones that do you should see a "E H/V Acc" field and two values like "35.5/23.3 ft" in the Status view.

I've added support for vertical accuracy to the issue for a future UI that focuses on what APIs/data a particular device supports at https://github.com/barbeau/gpstest/issues/313.

>+ or - for V error

This is likely the above issue - if your device doesn't support vertical accuracy, you won't see this on the graph in the Accuracy pane

>scale for grid

Not sure I understand this - could you elaborate?

>Your...(position pane)
in EDIT mode, the SAVE button is at the edge of the pane, leading to accidental taps on the grid (changing position!) rather than the Save button. Suggest moving the button to the white space near the top right corner of the pane.

Thanks for the feedback, I'll take it into account as I consider future UI changes. I have to balance ease of access too - anything you put towards the top of the screen is hard to tap when using with a single hand on most larger screen devices.

barbeau, 2020-12-07:
I think "scale for grid" means a map scale bar, which has an issue open at https://github.com/barbeau/gpstest/issues/395, so I'm going to close this out.

closed 2020-12-07 by barbeau