barbeau/gpstest #450: Support external NMEA source over TCP (e.g. RTKLIB, RTK-capable external receiver)

gdt, 2020-11-23:
There are ways to get GPS information into an Android phone from non-traditional sources; these are used to get higher accuracy, and in particular to support RTK.

Mechanisms include:
  * app that needs position makes a TCP connection to a socket on another computer (typically on hotspot LAN) or to localhost
  *  app that needs position listens on a  port and the remote device connects

Setups incluee
  * RTKLIB running on the phone, either with phone raw measurements or an external receiver
  * RTK-capable GPS receiver on another computer (e.g u-blox F9P)

A working example of this setup is Vespucci (see https://github.com/MarcusWolschon/osmeditor4android ) which has advanced setttings for location.

Vespucci's config is basically the right approach.   It might be nice to have "try to connect and if not use normal" but that might be confusing.  It might be nice to have the concept of  mulitple configurations and a quick dropdown to switch among them.

When out in the field with Vespucci and F9P, it would be great to turn on GPSTest and see the L1/L2 satellite list, RTK fix status, and various other things.

This is conceptually similar to #79  and I think conceptually distinct from #308.

gdt, 2021-06-03:
Hey bot: Despite no activity, this is still valid.

barbeau, 2023-12-19:
Not stale

gdt, 2025-01-24:
Hey stale see #645.

gdt, 2025-01-24:
Hey stale see #645.

closed 2026-02-28 by stale[bot]