barbeau/gpstest #455: Can google maps be replaced by other mapping software？

skittlz-lab, 2020-12-10:


barbeau, 2020-12-10:
Yes! Please check out the F-Droid release for non-Google Maps devices - https://f-droid.org/packages/com.android.gpstest.osmdroid/.

closed 2020-12-10 by barbeau

skittlz-lab, 2020-12-11:
but this one also uses Google Maps..:(

barbeau, 2020-12-11:
Nope, the F-Droid release uses [OSMDroid](https://github.com/osmdroid/osmdroid) with a combination of Mapbox and Maptiler tiles that are based on [OpenStreetMap](https://www.openstreetmap.org/) data.

skittlz-lab, 2020-12-16:
Ok thanks for your answer :)

skittlz-lab, 2020-12-16:
But the source code I downloaded and then imported into the phone from Android studio is wrong

skittlz-lab, 2020-12-16:
than still uses Google Maps

barbeau, 2020-12-16:
You'll want to build the osmdroid flavor - see https://github.com/barbeau/gpstest/blob/master/BUILD.MD#building-in-android-studio