barbeau/gpstest #456: Google incorrectly detecting background location use

barbeau, 2020-12-10:
**Describe the bug**
On submission of v3.9.0 to Google Play, Google kicked it back with this message triggered by their new location use screening:

>We detected that your app contains at least one feature that requests access to location in the background, however  your permission declaration form did not reflect this. Please log in to your Play Console to resubmit your location declaration form. You may either remove location in the background from your app or indicate that the usage is in the background.

GPSTest only works in the foreground and doesn't run in the background (hence https://github.com/barbeau/gpstest/issues/299 ), but I'm guessing that Google's automated system is looking for keywords within the app and flagged the old setting "Stop GNSS in background" may have triggered this. 

The first version of GPSTest dates back to Android 1.5 where if you didn't unregistered the LocationListener in onPause() GPS would continue running in the background. I left that behavior in place because some users saw that as a feature, not a bug. I did introduce at some point the setting to follow normal Android conventions where users could choose to unregister the LocationListener when the Activity was paused, which is the above setting.

However, in some Android version (maybe 7 or 8?) Android started automatically unregistering the LocationListener when apps were put into the background, regardless of what the developer did in their code (mainly to ensure apps weren't draining your battery). So on modern Android devices the above setting doesn't have any affect - no matter what, the app will always stop listening to GPS when put in the background.

So, regardless of the Google notice, this code should be updated to follow the modern Android conventions of unregistered the LocationListener in onPause().

**To Reproduce**
Steps to reproduce the behavior:
1. Submit app to Google Play

**Expected behavior**
App update should be accepted as the app doesn't run in the background

**Observed behavior**
App update was rejected with the above message

closed 2020-12-10 by barbeau

Rikk, 2020-12-13:
I thought this feature was useful when GPS takes a long time to detect my location and I need to use another app in the while.
Perhaps, this strategy didn't work and GPS doesn't stay working when app is in background (despite the message in notification bar)?

barbeau, 2020-12-14:
@Rikk Which device have you used this on, and which Android version? If the GPS icon says on in the notification bar then it could be having an effect.

I do have true background execution planned as a future feature in https://github.com/barbeau/gpstest/issues/299.

Rikk, 2020-12-15:
Android 5...
#299 will be very nice indeed.

barbeau, 2020-12-15:
@Rikk Yeah, Android 5 is old enough that you would see different behavior. If this is a critical thing for you you can always use older releases of GPSTest (v3.8.x or lower) downloaded from here:
https://github.com/barbeau/gpstest/releases

Otherwise #299 is something I plan to get to in the future.

commit ae93b7ff (2020-12-10): fix: Always stop LocationListener in activity onPause()

Closes #456