barbeau/gpstest #457: Wrong layout in share dialog

sm4rk0, 2020-12-13:
**Describe the bug**
Actions in the share dialog are drawn over the "Include altitude" checkbox.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. Tap the share button

**App, Device and Android version:** 

GPSTest 3.9.1
Samsung Galaxy S10e
Android version 10 (stock)
Google Play Services 20.47.13

**Screenshots:** 

[Google photos](https://photos.app.goo.gl/qXQnB3kjnwRSMjMHA), before and after checking the "Include altitude" checkbox.

naoliv, 2020-12-13:
Indeed:
[image]

It's also not possible to properly see the send button at the last share screen:

[image]

barbeau, 2020-12-14:
@sm4rk0 @naoliv Thanks for flagging this! Seems like the layout isn't working the same on all devices, then :(.

@naoliv Which device and Android version are your screenshots from?

naoliv, 2020-12-14:
Mine is a Mi A2 Lite, with Android 10.

sm4rk0, 2020-12-15:
If it helps, my font size setting is 3/8 and zoom is 1/5.

closed 2020-12-18 by barbeau

naoliv, 2020-12-19:
@barbeau the last screen still isn't right here:

[image]

If I select the text it gets resized to the proper size:

[image]

If I play with `Include altitute`:

[image]

It's also properly resized:

[image]

barbeau, 2020-12-19:
@naoliv Thanks! Could you open a new issue for that please?

commit ea1ddc42 (2020-12-18): fix: Ensure Share location buttons appear below altitude

Closes #457