barbeau/gpstest #458: Crash if orientation is changed while share dialog is open

sm4rk0, 2020-12-13:
**To Reproduce**
Steps to reproduce the behavior:
1. Open the app
2. Tap the share button
3. Change the phone orientation

**App, Device and Android version:** 
GPSTest 3.9.1
Samsung Galaxy S10e
Android version 10 (stock)
Google Play Services 20.47.13

barbeau, 2020-12-14:
@sm4rk0 Thanks for the report! Can you submit an error report via Google Play the next time this happens?

Also, does the crash happen when you have Share open and rotate the phone before getting a fix, after getting a fix, or both?

sm4rk0, 2020-12-15:
There's no share button before a fix.

Sorry, forgot to mention, I'm running an f-droid version and phone is not rooted (yet). Not sure how to get an error report/logcat. But can play with it and try to replicate. So far I can see that crash happens pretty randomly (un)related to the share dialog.

Saboe, 2020-12-15:
I can confirm this issue. But if I tap once or twice one of the buttons 'Decimal degrees', 'Log', or 'Device' and then turn the phone to landscape mode, GPStest will not crash. Just as a hint to fix it.

barbeau, 2020-12-15:
@Saboe Thanks for the info. Are you using the Google Play version of GPSTest or F-Droid? What device and Android version too?

Unfortunately I can't reproduce this on a Galaxy S8+ w/ Android 9 - it doesn't crash. 

If someone can reproduce this and submit an error report via the Google Play version (you'll be prompted after the crash) it would help me track down the issue.

Alternately you can download Android Studio and look at LogCat and share the stack trace that's shown there after the crash:
https://developer.android.com/studio/debug/am-logcat

Saboe, 2020-12-15:
I'm using the F-Droid version 3.9.1 on my Fairphone 3 with Android 10, build 8901.3.A.0054.20200929.
Btw, if the accuracy mode is running the behaviour of the app is very slow.

I suppose that installing Android Studio assumes that I've more ore less knowledge about Android SW-development, right? But I'm sorry, I'm not an Android developer and until now I've nothing done with it. Perhaps it is a solution to build a separately debug version which I (and other's) can install and run gpstest until it crashes. The debug report afterwards can be send to you for further investigation.

barbeau, 2020-12-16:
Ok, I think I was able to reproduce this, but I believe it only happens if you change orientation on the FDroid version while the map is active - meaning you have to either start with or switch to either the Map or Accuracy screen during app execution (and then optionally switch to another view like Status). For example, if the app starts up to Status and you don't switch to one of the map screens it doesn't happen.

If everyone could check if the above matches what you're seeing that would help.

sm4rk0, 2020-12-16:
Here's the logcat from one crash:
1. open the app
2. get a fix
3. move to map
4. change rotation

```12-16 02:09:00.861 30746 30746 D AndroidRuntime: Shutting down VM
--------- beginning of crash
12-16 02:09:00.861 30746 30746 E AndroidRuntime: FATAL EXCEPTION: main
12-16 02:09:00.861 30746 30746 E AndroidRuntime: Process: com.android.gpstest.osmdroid, PID: 30746
12-16 02:09:00.861 30746 30746 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'void org.osmdroid.views.overlay.LinearRing.setPoints(java.util.List)' on a null object reference
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at org.osmdroid.views.overlay.Polygon.setPoints(Polygon.java:159)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at com.android.gpstest.GpsMapFragment.onLocationChanged(GpsMapFragment.java:267)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at com.android.gpstest.GpsTestActivity.onLocationChanged(GpsTestActivity.java:1584)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:352)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport.access$000(LocationManager.java:275)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport$1.handleMessage(LocationManager.java:291)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:237)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:8167)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
12-16 02:09:00.861 30746 30746 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
```

and here's from another:
1. open the app in map view (remembered from previous run)
2. move to status view
3. change orientation
4. get a fix

```12-16 02:19:29.720 15542 15542 D AndroidRuntime: Shutting down VM
12-16 02:19:29.720 15542 15542 E AndroidRuntime: FATAL EXCEPTION: main
12-16 02:19:29.720 15542 15542 E AndroidRuntime: Process: com.android.gpstest.osmdroid, PID: 15542
12-16 02:19:29.720 15542 15542 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.Context android.view.ViewGroup.getContext()' on a null object reference
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at org.osmdroid.views.MapViewRepository.getDefaultMarkerIcon(MapViewRepository.java:74)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at org.osmdroid.views.overlay.Marker.setDefaultIcon(Marker.java:139)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at org.osmdroid.views.overlay.Marker.<init>(Marker.java:114)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at org.osmdroid.views.overlay.Marker.<init>(Marker.java:92)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at com.android.gpstest.GpsMapFragment.onLocationChanged(GpsMapFragment.java:304)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at com.android.gpstest.GpsTestActivity.onLocationChanged(GpsTestActivity.java:1584)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport._handleMessage(LocationManager.java:352)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport.access$000(LocationManager.java:275)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.location.LocationManager$ListenerTransport$1.handleMessage(LocationManager.java:291)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:107)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:237)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:8167)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
12-16 02:19:29.720 15542 15542 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)```

barbeau, 2020-12-16:
If I update GPSTest to the latest version of OSMDroid (6.1.8) and just try to open the app to the Map I get this crash:

```
2020-12-16 12:06:51.928 1950-1950/com.android.gpstest.osmdroid E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.gpstest.osmdroid, PID: 1950
    java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String java.io.File.getAbsolutePath()' on a null object reference
        at org.osmdroid.config.DefaultConfigurationProvider.load(DefaultConfigurationProvider.java:337)
        at com.android.gpstest.GpsMapFragment.onCreateView(GpsMapFragment.java:103)
        at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2600)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:881)
        at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1303)
        at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:439)
        at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2079)
        at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1869)
        at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1824)
        at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1727)
        at androidx.fragment.app.FragmentManagerImpl.dispatchStateChange(FragmentManagerImpl.java:2663)
        at androidx.fragment.app.FragmentManagerImpl.dispatchActivityCreated(FragmentManagerImpl.java:2613)
        at androidx.fragment.app.FragmentController.dispatchActivityCreated(FragmentController.java:246)
        at androidx.fragment.app.FragmentActivity.onStart(FragmentActivity.java:542)
        at androidx.appcompat.app.AppCompatActivity.onStart(AppCompatActivity.java:201)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1391)
        at android.app.Activity.performStart(Activity.java:7348)
        at android.app.ActivityThread.handleStartActivity(ActivityThread.java:3140)
        at android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:180)
        at android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:165)
        at android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:142)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:70)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1950)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7073)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:494)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

```

...which seems to be this bug in OSMDroid:
https://github.com/osmdroid/osmdroid/issues/1636

barbeau, 2020-12-17:
Thanks for the info everyone! It should be fixed in the next release.

commit 5e4d2167 (2020-12-17): fix: Check if OSMDroid is laid out before updating

Also bump OSMDroid version

Closes #458
closed 2020-12-17 by barbeau