barbeau/gpstest #459: gpstest crashes after a while when in accuracy mode

Saboe, 2020-12-15:
When I start the accuracy mode everything seems to be OK so far but after a while, mostly between 15 and 40 seconds GPStest crashes. The time until crash differs from time to time, I've no clue where it depends on.

My phone is a Fairphone 3 with Android 10.

barbeau, 2020-12-15:
@Saboe Thanks for the report! Is this using the version from F-Droid or Google Play? Are you using app version v3.9.1, and if so is it new to the app version v3.9.1?

Saboe, 2020-12-15:
I'm using the F-Droid version 3.9.1 on my Fairphone 3 with Android 10, build 8901.3.A.0054.20200929.

This bug is **not** new to this version. I've experienced this bug several month ago (but took no action to report this).

barbeau, 2020-12-16:
@Saboe ok, thanks! I think this is a duplicate of https://github.com/barbeau/gpstest/issues/458, so I'm going to close this and will post more to the other issue.

closed 2020-12-16 by barbeau