barbeau/gpstest #463: Share menu crashes

mirabilos, 2020-12-19:
**Describe the bug**

Three dots menu → Share opens a dialogue that GPSTest has crashed

**To Reproduce**
Steps to reproduce the behavior:
1. Start GPSTest
2. Wait for a fix
3. Try to share the infos for the developer
4. See the crash dialogue

**Expected behavior**

I see some window where I can upload device data, as on the newer Android

**Observed behavior**

Crash

**App, Device and Android version:** 

There’s no Help→About, but I just updated this in F-Droid to latest.

Android 4.4 (LiquidSmooth 3.2), HTC Desire “bravo”, MicroG

barbeau, 2020-12-19:
Thanks for the report @mirabilos, I can produce this on an Android 4.4 emulator. It's new in v3.9, and here's the stacktrace:

~~~
12-18 19:44:28.453 6660-6660/com.android.gpstest E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.gpstest, PID: 6660
    android.content.res.Resources$NotFoundException: File res/drawable/ic_baseline_location_on_24.xml from drawable resource ID #0x7f08009b. If the resource you are trying to use is a vector resource, you may be referencing it in an unsupported way. See AppCompatDelegate.setCompatVectorFromResourcesEnabled() for more info.
        at android.content.res.Resources.loadDrawable(Resources.java:2101)
        at android.content.res.Resources.getDrawable(Resources.java:700)
        at androidx.core.content.ContextCompat.getDrawable(ContextCompat.java:457)
        at com.android.gpstest.dialog.ShareDialogFragment$onCreateDialog$1.onConfigureTab(ShareDialogFragment.kt:62)
        at com.google.android.material.tabs.TabLayoutMediator.populateTabsFromPagerAdapter(TabLayoutMediator.java:170)
        at com.google.android.material.tabs.TabLayoutMediator.attach(TabLayoutMediator.java:138)
        at com.android.gpstest.dialog.ShareDialogFragment.onCreateDialog(ShareDialogFragment.kt:73)
        at androidx.fragment.app.DialogFragment.onGetLayoutInflater(DialogFragment.java:380)
        at androidx.fragment.app.Fragment.performGetLayoutInflater(Fragment.java:1412)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:881)
        at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238)
        at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:1303)
        at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:439)
        at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2079)
        at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1869)
        at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1824)
        at androidx.fragment.app.FragmentManagerImpl.execPendingActions(FragmentManagerImpl.java:1727)
        at androidx.fragment.app.FragmentManagerImpl$2.run(FragmentManagerImpl.java:150)
        at android.os.Handler.handleCallback(Handler.java:733)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:136)
        at android.app.ActivityThread.main(ActivityThread.java:5017)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:515)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:779)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:595)
        at dalvik.system.NativeStart.main(Native Method)
     Caused by: org.xmlpull.v1.XmlPullParserException: Binary XML file line #1: invalid drawable tag vector
        at android.graphics.drawable.Drawable.createFromXmlInner(Drawable.java:933)
        at android.graphics.drawable.Drawable.createFromXml(Drawable.java:877)
        at android.content.res.Resources.loadDrawable(Resources.java:2097)
~~~

I'll take a look.

mirabilos, 2020-12-19:
Sean Barbeau dixit:

>Thanks for the report @mirabilos, I can produce this on an Android 4.4
>emulator.

Ah, thanks the gods. For some reason, I couldn’t get the device to
show up in dmesg so I couldn’t adb in to get a logcat, then had to
go.

>I'll take a look.

Thanks!

closed 2020-12-24 by barbeau

barbeau, 2021-01-05:
This fix is released in v3.9.4 if you want to try it out.

mirabilos, 2021-01-08:
Sean Barbeau dixit:

>This fix is released in v3.9.4 if you want to try it out.

It has yet to show up in F-Droid as update, but sure once it does.

commit 1f868b88 (2020-12-24): fix: Reference drawable directly for ViewPager2

* Also bump AppCompat version

Closes #463