barbeau/gpstest #464: Small issue with the share device layout

naoliv, 2020-12-19:
Now with version `3.9.3` I am seeing this (on a Xiaomi Mi A2 Lite, Android 10):

[image]

The "send" button isn't fully displayed.
But if I select the text it gets resized to the proper size:

[image]

If I play with `Include altitute`:

[image]

It's also properly resized:

[image]

Thank you!

barbeau, 2021-01-20:
@naoliv Thanks for the report. I can't seem to reproduce this. Does it always happen on your device or just sometimes? Can you give me an exact set of steps to reproduce it? Could you reproduce it on another device or emulator?

naoliv, 2021-01-20:
It always happens here.
I open GPSTest, wait for a fix, share button, `DEVICE` icon and I can always see 
[image]

Right now I don't have a computer to run an emulator nor another device to test, but I can give you a feedback as soon as I am able to test it.

JonnyTech, 2021-01-21:
I can confirm with v3.9.5, although as soon as I touch anywhere with text (ie sentence starting `Uploads your device`) then the box expands correctly:

[image]
[image]

Steps:
- start app
- wait for location fix
- click share icon
- click device icon
-- cropped box shown
- click text
-- box expands correctly

naoliv, 2021-06-03:
It's not stale.
Still happens on `3.9.11`.