barbeau/gpstest #469: Share button kills app

JonnyTech, 2020-12-31:
**Describe the bug**
Pressing the share button instantly kills the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Start GPStest
2. Wait for location fix
3. Press share
4. App force closes

**Expected behavior**
Share dislog to appear

**Observed behavior**
App closes and launcher shown

**App, Device and Android version:** 

What version of GPSTest did you encounter this on (see "Help->About")?
V3.9.3 (18080 osmdroid) from F-Droid
GNSS HW year 2016
Platform 10
API level 29

What make and model device (e.g., Samsung Galaxy S10) did you encounter this on?
Xiaomi Redmi Go (tiare)

What Android version (e.g., Android 10.0) are you running?
AOSP 10

Is it the stock version from the manufacturer or a custom ROM?
LineageOS 17.1 GSI

What version of Google Play Services is running on your device?
None - using microG

barbeau, 2020-12-31:
@JonnyTech Thanks for the report! I'm guessing this is a duplicate of https://github.com/barbeau/gpstest/issues/463, which I've fixed in the main branch but haven't pushed out in a release yet due to the holidays. I'm going to close this out but if you continue to see the issue in the next release please reopen.

barbeau, 2021-01-05:
@JonnyTech v3.9.4 is now out with the fix for #463 - could you give this a shot and see if it fixes the issue you're seeing too?

JonnyTech, 2021-01-05:
I notice that your osmdroid release is signed with a different key to the fdroid one so had to do the uninstall/reinstall dance. But sadly, no, with regret I have to inform you that the share button still kills the app. Is there anything that I can provide you with to help diagnose the bug?

JonnyTech, 2021-01-05:
@barbeau can you reopen this issue? I am unable to.

JonnyTech, 2021-01-05:
I do not have Android Studio installed (currently in the middle of a project so I cannot install it) but can provide a `strace` from a root shell if that helps.
```
# am start -n com.android.gpstest.osmdroid/com.android.gpstest.GpsTestActivity
Starting: Intent { cmp=com.android.gpstest.osmdroid/com.android.gpstest.GpsTestActivity }
# pidof com.android.gpstest.osmdroid
3299
# strace -p 3299 -o gpsteststrace.txt
```
[gpsteststrace.txt](https://github.com/barbeau/gpstest/files/5771996/gpsteststrace.txt)

Here is a logcat from this command: `adb logcat AndroidRuntime:E *:S`: [logcatruntime.txt](https://github.com/barbeau/gpstest/files/5772020/logcatruntime.txt)

And a `dmesg` too: [dmesg.txt](https://github.com/barbeau/gpstest/files/5772067/dmesg.txt)

Hope that helps.

barbeau, 2021-01-06:
@JonnyTech Thanks!! The logcatruntime.txt has what I needed:

~~~
01-05 18:52:18.244  3540  3540 E AndroidRuntime: FATAL EXCEPTION: main
01-05 18:52:18.244  3540  3540 E AndroidRuntime: Process: com.android.gpstest.osmdroid, PID: 3540
01-05 18:52:18.244  3540  3540 E AndroidRuntime: java.lang.RuntimeException: java.lang.reflect.InvocationTargetException
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:501)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:940)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: Caused by: java.lang.reflect.InvocationTargetException
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:491)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	... 1 more
01-05 18:52:18.244  3540  3540 E AndroidRuntime: Caused by: java.io.IOException: no result
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.location.Geocoder.getFromLocation(Geocoder.java:136)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at com.android.gpstest.dialog.UploadDeviceInfoFragment.onViewCreated(UploadDeviceInfoFragment.kt:66)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.moveToState(FragmentManagerImpl.java:892)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.moveFragmentToExpectedState(FragmentManagerImpl.java:1238)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.BackStackRecord.executeOps(BackStackRecord.java:434)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.executeOps(FragmentManagerImpl.java:2079)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.executeOpsTogether(FragmentManagerImpl.java:1869)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.removeRedundantOperationsAndExecute(FragmentManagerImpl.java:1824)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.FragmentManagerImpl.execSingleAction(FragmentManagerImpl.java:1696)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.fragment.app.BackStackRecord.commitNow(BackStackRecord.java:293)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.placeFragmentInViewHolder(FragmentStateAdapter.java:341)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:276)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:67)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView.dispatchChildAttached(RecyclerView.java:7556)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$5.addView(RecyclerView.java:860)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.ChildHelper.addView(ChildHelper.java:107)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addViewInt(RecyclerView.java:8601)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8559)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8547)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:1641)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:1587)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.LinearLayoutManager.onLayoutChildren(LinearLayoutManager.java:665)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView.dispatchLayoutStep2(RecyclerView.java:4134)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView.dispatchLayout(RecyclerView.java:3857)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView.onLayout(RecyclerView.java:4404)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.View.layout(View.java:22466)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6304)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.viewpager2.widget.ViewPager2.onLayout(ViewPager2.java:527)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.View.layout(View.java:22466)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6304)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at androidx.constraintlayout.widget.ConstraintLayout.onLayout(ConstraintLayout.java:1762)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.View.layout(View.java:22466)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6304)
[... cut ...]
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:100)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:214)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	at android.app.ActivityThread.main(ActivityThread.java:7356)
01-05 18:52:18.244  3540  3540 E AndroidRuntime: 	... 3 more
~~~

Looks like some AOSP devices that don't implement the Android geocoder throw an IOException instead of just returning an empty list of results. I'll try to get another release out with a fix soon.

closed 2021-01-06 by barbeau

barbeau, 2021-01-06:
v3.9.5 is out now that should hopefully fix this:
https://github.com/barbeau/gpstest/releases/tag/v3.9.5

JonnyTech, 2021-01-06:
Perfect - sharing now works again - thank you very much for your efforts.

JonnyTech, 2021-01-14:
For completeness, confirming that the F-Droid build is fixed too.

commit 46a8fd9e (2021-01-06): fix: Handle exceptions and nulls when geocoding using Android API

Looks like some AOSP devices that don't implement the Android geocoder say it's supported via Geocoder.isPresent() but then throw an IOException instead of just returning an empty list of results. The Android API docs also say that getFromLocation() can return null, so let's handle that too.

Closes #469