barbeau/gpstest #471: There are two features I'm interested in that I hope can be added.

Jia-le-wang, 2021-01-07:
I've been following GPSTEST for a long time, and I have to admit, this software works really well.It can show the azimuth of the satellite, the constellations tracked, signal quality,TTFF and so on. However, this is not enough for a deeper understanding and research of the high accuracy positioning performance of smart phones.

1. TTFF is heavily influenced by GNSS auxiliary data, including A-GNSS server, extended ephemeris, and hot start data.So, is it possible to add a function to continuously test TTFF of standalone GNSS ?I mean, make sure the phone is positioned and record the TTFF multiple times without interference from assisted data.

2. Could you add a feature to support the output of raw satellite observations dara of RINEX 3.03+?Because this is necessary for high precision GNSS positioning and will make the software function more perfect.I have noticed that GNSSLogger V3.0 can now output observations in RINEX 3.03 [format.](https://play.google.com/store/apps/details?id=com.google.android.apps.location.gps.gnsslogger&hl=en_US&gl=US)However, it still has many imperfections, such as formatting errors, carrier phase recording errors, missing pseudorange values, and so on.

barbeau, 2021-01-07:
> TTFF is heavily influenced by GNSS auxiliary data, including A-GNSS server, extended ephemeris, and hot start data.So, is it possible to add a function to continuously test TTFF of standalone GNSS [...]

Unfortunately no - Android APIs don't given enough access to apps to understand how and when assistance data is being used.

>Could you add a feature to support the output of raw satellite observations dara of RINEX 3.03+?Because this is necessary for high precision GNSS positioning and will make the software function more perfect.I have noticed that GNSSLogger V3.0 can now output observations in RINEX 3.03 format.However, it still has many imperfections, such as formatting errors, carrier phase recording errors, missing part of the satellite pseudorange values, and so on.

This is certainly possible! I'm not very familiar with the RINEX format myself, so any help from others that want to take this on is welcome. Especially if someone knows of an existing OSS library that could help with the heavy-lifting.

gdt, 2021-01-11:
I'm not barbeau@ but if I were I'd ask that this be closed and two separate issues be opened, each with a summary of what is being asked for in the title.   Putting two independent things in one issue just doesn't work with how trackers are used.

barbeau, 2021-01-11:
>Putting two independent things in one issue just doesn't work with how trackers are used.

💯 

>I'd ask that this be closed and two separate issues be opened, each with a summary of what is being asked for in the title

@gdt I usually would, but given the first issue mentioned isn't viable I just changed the title and co-opted this issue for the 2nd RINEX format issue, so we can track that here going forward.

barbeau, 2021-02-11:
Note that this tool supports converting from the CSV output of GPSTest to RINEX:
https://github.com/rokubun/android_rinex

I think this should effectively solve this issue, although I'd be interested to know if anyone sees significant value in having direct RINEX output support within GPSTest.

vkbellis, 2021-04-19:
Is this the same thing that GnssLogger is using to spit out its RINEX?

I've yet to test GPSTest's CSV run through Alex's RINEX writer; however, I am curious about its formatting compliance, and what, if any, format checking tool(s) were used in any evaluation of the produced file.

barbeau, 2021-04-19:
>Is this the same thing that GnssLogger is using to spit out its RINEX?

No, Google has their own code in GnssLogger to directly output to RINEX from the app. However, it's not open-source, so it's not something I can port over to GPSTest. Google [has stopped doing open-source development on GnssLogger](https://github.com/google/gps-measurement-tools/pull/21#issuecomment-756215993), so the version on Google Play now with RINEX export is a closed source fork that they are working on internally.

https://github.com/rokubun/android_rinex is a desktop tool that you can use to convert the CSV file output from GnssLogger or GPSTest to RINEX. It was developed before GnssLogger added direct RINEX output.

ZiglioUK, 2022-02-21:
Here's an open source implementation in Java if you're still interested:
https://github.com/goGPS-Project/goGPS_Java/tree/master/src/main/java/org/gogpsproject/producer/rinex

barbeau, 2022-02-25:
Thanks @ZiglioUK! Looks like that code is licensed under L-GPL which makes it trickier to integrate into GPSTest. If you stumble across anything under Apache v2 or MIT please let me know.

ZiglioUK, 2022-09-14:
Sorry @barbeau , I was meant to ask the oweners of GoGPS whether alternative licensing could be considered. I will, if you're still interested

barbeau, 2022-09-14:
@ZiglioUK thanks! That would be helpful.

gdt, 2023-04-03:
Hey bot, still valid.

ZiglioUK, 2023-04-03:
Hi there, back in 2022 I did ask @ege010 (the owner of GoGPS) whether we could re-release the library under a different or as an additional license and he did not have any objections. I will ping him again. I don't really know the difference between LGPL, Apache v2 or MIT so I wouldn't know whether they're mutually exclusive for example.

ege010, 2023-04-14:
Hi, indeed I do not have objections to re-releasing the library under a different license. Directions/suggestions on the proper way to do that would be appreciated. I suppose first of all we would have to contact all contributors and check whether they agree.

barbeau, 2023-04-14:
@ege010 Thanks for being willing to do this, and thanks @ZiglioUK for reaching out to @ege010.

Before you go to the effort of reaching out to contributors and re-releasing under a new license, I'm also looking at another option for adding RINEX support. I'll report back later on this option, although it may take some time.

barbeau, 2023-11-01:
Not stale

closed 2024-05-22 by stale[bot]