barbeau/gpstest #480: GLONASS isn't working on the LOS14.1 using the serranoltexx device

The-First-King, 2021-02-02:
I'm using 2 devices: the crackling running LOS16.1 and the serranoltexx running LOS14.1. The problem is present on the serranoltexx only. Not sure that this is gpstest related issue. The most probably it's a device settings. But I would be very happy if anyone could help to narrow down the problem.

https://wiki.lineageos.org/devices/serranoltexx/

According to the spec data serranoltexx GPS chip supports GLONASS. With the following content of the gps.conf file the device can perform GPS fix within 5-15 seconds. Before that it has been necessary to wait for 5 minutes approximately. 
```
XTRA_SERVER_1=http://xtrapath1.izatcloud.net/xtra2.bin
XTRA_SERVER_2=http://xtrapath2.izatcloud.net/xtra2.bin
XTRA_SERVER_3=http://xtrapath3.izatcloud.net/xtra2.bin
XTRA_TEST_ENABLED = 1
XTRA_VERSION_CHECK=1
ERR_ESTIMATE=0
NTP_SERVER=pool.ntp.org
XTRA_CA_PATH=/system/etc/security/cacerts
DEBUG_LEVEL=1
INTERMEDIATE_POS=0
SUPL_VER=0x10000
SUPL_ES=1
USE_EMERGENCY_PDN_FOR_EMERGENCY_SUPL=1
CAPABILITIES=0x33
SUPL_HOST=supl.google.com
SUPL_PORT=7276
LPP_PROFILE=0
NMEA_PROVIDER=0
SGLTE_TARGET=0
A_GLONASS_POS_PROTOCOL_SELECT=0
```
Thank you in advance.

The-First-King, 2021-02-02:
On the XDA forum I have found it mentioned that it's an android issue. In Oreo and Nougat no GLONASS fix due to change in software. Only with Marshmallow it was working fine. And now I can assume that Pie also have a fix. Because as I mentioned the crackling device works great.

Interesting if I have root access can I fix it manually if this is true an android problem?

The-First-King, 2021-02-03:
Based on information in another forum it depends on the Android platform implementation. The default framework that Google provides does not carry information for any constellation except GPS. It means that the device could be using GLONASS and just not see the satellites show up because the Framework has no way to pass this info from the hardware device driver to the Framework.

[image]

What does this screenshot mean? Is it looks like GLONASS satellites are in use?

barbeau, 2021-02-04:
@The-First-King Android first officially introduced support for non-GPS satellites in Android 7 (Nougat). Prior to that, some OEMs represented GLONASS using PRNs between 65 and 96, which GPSTest includes legacy support for:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/SatelliteUtils.java#L79

Beyond that, though, if you're not running official OEM firmware it's going to be hit or miss getting this to work. The hardware, drivers, and Android all have to be in agreement on the data being passed through in order for it to show up correctly in the Android APIs and therefore Android apps, and that's going to be hard if they are coming from different sources.

In the above screenshot, none of the GLONASS satellites are in use - if there were used, it would have a "U" flag. Also the device doesn't seem to seeing any GLONASS satellite signals either - the C/N0 values are all blank.

Also, the calculated GPS time seems wrong because it has a red background - see this article for more details:
https://barbeau.medium.com/how-to-detect-gps-week-rollover-problems-on-android-5cc739f2fa9c

I'm going to close this issue because it's not related to GPSTest code specifically, but feel free to continue to post more as you find out how to get this to work, as it could be valuable to others.

closed 2021-02-04 by barbeau

The-First-King, 2021-02-05:
Thank you for detailed explanation. I did a few more tests and I found that GPS chip can see GLONASS satellites. Howewer I do not know why it can not get a fix. I found the fix is only possible to get with GPS satellites. Even the combination of  GLONASS with GPS is not possible on this device. Now I'm trying to understand if I can get it repaired manually or not by changing settings in the gps config file or replacing the chip driver.

[image]

GPS week rollover problem is indeed present on my device. But I found it's not relevant to connection problems or coordinates accuracy. 

[image]

The-First-King, 2024-01-24:
This problem is not present in LineageOS 18.1 for S4 Mini.