barbeau/gpstest #482: Forward GNSS data direct via network to an IP/TCP/UDP endpoint

mbehn1976, 2021-02-07:
I was wondering if there is any plan to add a feature to send the data (RAW and/or NMEA) direct via network to an IP/TCP/UDP endpoint as well, e.g. in addition or alternative to local logging. I would for example like to remotely record the data with the U-Center from U-Blox, but locally use GPS-Test.

barbeau, 2021-02-10:
@mbehn1976 Thanks for the feedback! It's not something I currently have planned but it might not be too difficult to implement given the current structure of the app, and contributions for features like this are always welcome.

Do you have more detailed connection information and specs for how exactly the message and connection would need to be formatted/setup for this to work? Is U-Center freely available to test with?

mbehn1976, 2021-02-12:
Hi Sean, the U-Center is freely available from https://www.u-blox.com/en/product/u-center - I have it running on Windows and Wine under Linux. U-Blox is a major GNSS chipset manufacturer out of Switzerland. The software allows for receiving remote GNSS traffic via Network - quite straight forward - tested it with the NEO6M U-blox on a ESP32 redirecting from local serial port to the remote IP. U-Center supports the NMEA 0183 and U-Blox own format  (see attached info from current manual https://www.u-blox.com/sites/default/files/u-center_Userguide_%28UBX-13005250%29.pdf ). There may be also other software out there which may support other formats.
[image]
[image]

barbeau, 2021-03-04:
Relevant u-center documentation:
* u-center User guide - https://www.u-blox.com/sites/default/files/u-center_Userguide_%28UBX-13005250%29.pdf
* u-blox Test server configuration (for cellular service, but similar protocol may be used for desktop) - https://www.u-blox.com/sites/default/files/products/documents/TestServerForSocketOperations_AppNote_%28UBX-14005690%29.pdf
* Connecting emlid GPS device to u-center:
    * Docs - https://docs.emlid.com/navio/common/dev/gps-ublox-ucenter/
    * Repo with software - https://github.com/emlid/Navio2
*  Connecting raspberry Pi to U-Center
    * https://ozzmaker.com/using-u-center-to-connect-to-a-raspberry-pi-with-an-attached-berrygps-imu-gsm/
    * https://www.raspberrypi.org/forums/viewtopic.php?t=200563

Java TCP library that works on Android - https://github.com/EsotericSoftware/kryonet

gdt, 2021-06-03:
This could be a good idea, but I think sending GPS to remote is something that needs to work in the backgrond, wtih the screen off (although GPS would still be on).

mbehn1976, 2022-04-25:
Would still be a desired function to compare different devices at one location.

monkeyfork, 2022-12-04:
i've searched extensively for this feature (gps and sensor data forward to udp endpoint on local network).  For me this would be for kind of wearable iot devices based on microcontrollers most raspberry pi variants or esp32s.  For wearable scenario i use android phone to get internet and wifi-capable micros for all else.  makes sense to use android device to get gps but damn! is it hard to do in android without tons of experience.  i'm just learning kotlin now and it seems to be excruciatingly hard to find simple gps or udp-server full examples to learn by.    expect coffee as soon as i figure how to do it.

barbeau, 2022-12-04:
@monkeyfork you might want to check out this article I wrote, which has links to two complete simpler GPS examples on GitHub:
https://link.medium.com/SHAkzWXFuvb

The Kotlin flow technique discussed in this article is implemented in GPSTest.

monkeyfork, 2022-12-04:
Thanks Sean, checking now.  Android Studio is still quite a mysterious
beast to me.  For iot stuff i've been using C forever, android studio and
kotlin is quite a hurdle to jump.
thx,
doug jones

barbeau, 2024-01-12:
Not stale

ejrobinson, 2024-03-08:
I'd love to see this implemented, if you can forward the full RAW data over TCP/UDB you can hook it into RTKLIB or similar, even running on the same device, which opens up loads of possibilities for RTK on Android devices.

mbehn1976, 2024-10-06:
Indeed this is what I also would like to try.

pauliojanpera, 2025-11-09:
Here is a Termux script based implementation of GPSTest log sharing for consumption by Geoclue on a computer attached to the phone hotspot. I of course wish that this could serve as inspiration for a native implementation in GPSTest.

https://github.com/OpenOrienteering/mapper/issues/1569#issuecomment-3507418668