barbeau/gpstest #485: NMEA Sentences is not received

ChamichApps, 2021-02-19:
Given an Android device without Google Play Service Installed on it.
When I am running the GPTest application on the given device I am not receiving any NMEA Sentences, can you please tell me is this related to the fact that I don't have Google Play Service installed, or it's a bug?

From what I can see while debugging the application the NmeaListener is successfully registered but afterwords when the application starts working it's not receiving any messages.

barbeau, 2021-02-19:
The lack of Google Play Services shouldn't impact the functionality of receiving NMEA messages as the NMEA listener is part of the core Android framework.

That being said, if you have a device running a custom ROM or have modified it so it's no longer the OEM ROM running, this could certainly break NMEA messages, as they come from the chipset.

What make and model device is it, and is it modified from the OEM OS state? And are you running the F-Droid or Google build flavor of GPSTest?

Depending on the Android version you could also try toggling  the GPSTest setting "Use GNSS APIs". There are actually two NMEA listeners, the first one being the legacy one:
* https://developer.android.com/reference/android/location/GpsStatus.NmeaListener
* https://developer.android.com/reference/android/location/OnNmeaMessageListener

...and the above setting switches between then on devices with Android API 24 and higher.

barbeau, 2021-02-22:
Closing as part of project cleanup, as this doesn't seem to be related to the app. Please feel free to continue commenting and I'll provide whatever info I can.

closed 2021-02-22 by barbeau

ChamichApps, 2021-02-22:
Hi @barbeau Thanks for your quick response and sorry for the delay.
Yes look like this is not application related issue, but related to my hardware. Thanks a lot for your support and clarification