barbeau/gpstest #494: Database - Redmi Note 9 NFC mislabeled as Redmi 10X 4G

barbeau, 2021-04-06:
**Describe the bug**
Users reports a device labeled by the script as the  Redmi 10X 4G is actually a Redmi Note 9 NFC M2003J15SG (merlinnfc).

Looks like Redmi doesn't differentiate between several models based just on the MODEL value, as they are both listed as `M2003J15SC` in the Google CSV file as they are nearly the same hardware minus the NFC:

```
Redmi    Redmi Note 9     merlinnfc    M2003J15SC
Redmi    Redmi Note 9     merlin         M2003J15SC
Redmi    Redmi Note 9     merlinnfc    M2003J15SC
Redmi    Redmi 10X 4G   merlin         M2003J15SC
```

One option is to process the GPSTest CSV first, and dual-label these devices as "Redmi Note 9 / Redmi 10X 4G"

Reporting the DEVICE value (https://developer.android.com/reference/android/os/Build#DEVICE) probably wouldn't hurt too (although in this case it doesn't help).

It looks like Google is trying to fix this type of issue in Android S developer preview because they have added a few new "SKU" fields that could be used to better differentiate variants:
https://developer.android.com/reference/android/os/Build#ODM_SKU

**To Reproduce**
Steps to reproduce the behavior:
1. Share device info from Redmi Note 9 NFC 

**Expected behavior**
It gets labeled with the Name in the database of Redmi Note 9 NFC

**Observed behavior**
It should actually have a model number of `M2003J15SG`, but it reports the value `M2003J15SC`, so it gets labeled as Redmi 10X 4G from the Google CSV.

**App, Device and Android version:** 

Redmi Note 9 NFC M2003J15SG

**Screenshots:** 

Can be created by pressing the Volume Down and Power Button at the same time on Android 4.0 and higher.

pm-cz, 2021-04-06:
It is possible to obtain the human-readable name on the impacted device (and other new Xiaomi/POCO devices) from `ro.product.marketname` system property by executing `/system/bin/getprop ro.product.marketname`, but this is not a systematic solution, as it won't work on most other devices (the property is empty there).

Besides processing the GPSTest CSV first, it would be also possible to process Google CSV first and if multiple lines with records are found (there are 263+ devices with more than two names, but more than 1400 devices with at least two names):
- either let the user choose the submission device name from a list (this would require more interaction)
- or provide the joint name merged from the alternatives.

closed 2021-04-08 by barbeau

pm-cz, 2021-04-09:
Should the result be immediate after upload? Because I have tested the change and the new row in spreadsheet does not contain any device name at all...

barbeau, 2021-04-09:
@pm-cz No, the device names get populated by a script that runs about once an hour. For details see https://barbeau.medium.com/implementing-the-gpstest-database-a-serverless-architecture-c61a2084965.

commit 84515c27 (2021-04-08): chore: Add dual label for Note 9 / 10X 4G

Closes #494
commit 46eab253 (2021-04-08): chore: Script - Check GPSTest file before Google file

Closes #494