barbeau/gpstest #495: Can't upload to database on some devices

barbeau, 2021-04-09:
**Describe the bug**
On some devices when going to "Share->Device" it won't show the upload button and instead is stuck on the ”Go to Status and wait for a fix" text.

Maybe this is due to the fragment contents not being refreshed when visiting it multiple times? Some devices may cache the Share dialog fragments more aggressively than others. On all devices I've tested the dialog fragments are always recreated, but other Android variants could treat them differently.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a GNSS fix
1. Go to "Share->Device"

**Expected behavior**
I should be able to upload device details to Database.

**Observed behavior**
Instead it's stuck at the ”Go to Status and wait for a fix" text.

**App, Device and Android version:** 

Samsung J3 2016 (Android 7.1.2) no Share button.
Wiko Lenny 3 (Android 6.0) has Share button.
Nokia 3.1 (Android 10) has Share button.

All of the above never get shown the Upload option

**Screenshots:** 

[image]

skinkie, 2021-04-09:
Issue resolved in the F-droid version on Nokia 3.1.

barbeau, 2021-04-09:
Thanks for testing @skinkie! I'll push the fix I provided you to test after I clean it up a bit. It does seem like the issue is more aggressive fragment caching on some Android variants.

skinkie, 2021-04-11:
@barbeau I don't know if it is intentional. But on the same device, I am not capable of receiving the submit box again. There is a fix, and internet is connected.

barbeau, 2021-04-11:
Hmmm - no. In the current release, if you kill the app and then start it on the Status screen, and wait for a GNSS fix (lock icon to appear), then do Share->Device, you should always see the upload button (cloud icon with arrow). If you've already uploaded and no device capabilities have changed since last upload, when you tap on the upload button you should see a Toast popup saying that nothing has changed.

@skinkie The version I gave you to test cleans up some cases where older versions of the dialog telling you to wait for a fix could be cached (depending on Android system config) and not reflect the latest fix info, like if you tap on Share before getting a fix. In this version in all cases if you are on Status screen and see the lock icon, if you go to Share->Device you should see the cloud upload button.

barbeau, 2021-04-13:
Ok, I think I figured out what's going on here. In addition to the view state being cached longer than it should have, Android apparently doesn't allow new arguments to be passed to existing Fragments after they have been instantiated and attached to an Activity, so a new location being passed to an existing Fragment via `setArguments()` wouldn't be seen anyway:
https://developer.android.com/reference/androidx/fragment/app/Fragment#setArguments(android.os.Bundle)

So it seems the best solution here is to avoid used a cached DialoFragment entirely and just create a new one each time the user taps on the "Share" button.

barbeau, 2021-04-13:
A new Google Play beta version 3.9.11 will be rolling out shortly with these changes, and you can download the F-Droid version from the Releases page:
https://github.com/barbeau/gpstest/releases/tag/v3.9.11

Please test and let me know what you see!

Here's how to sign up for the beta:
https://play.google.com/apps/testing/com.android.gpstest

barbeau, 2021-04-22:
Re-opening, user is reporting this is still a problem at:
https://groups.google.com/g/gpstest_android/c/2fMe9sLp5y4/m/K0YVVCavAwAJ

...on:

>Phone is Honor 20 Lite, Model HRY-LX1T, with Huawei's EMUI (Android clone) identifying as 10.0.0.248(C431E8R1P7).  Processor Hisilicon Kirin 710F.

gb-dursley, 2021-06-15:
My Acer Iconia One 10 B3-A40 tablet (no phone h/w) Android 7.0, Kernel v.4.4.22 has this bug.  In isolation, not worth you bothering with - minority interest only!  But if it provides a clue that helps...

barbeau, 2021-09-20:
Reported on another device, LG K40 (LM-X420PR). Note the blank TTFF value, even though the lock icon is shown and a lat/lon value is shown. 

I'm guessing that this issue is caused on some devices by the device not reporting TTFF correctly, which causes issues in the upload, because GPSTest currently forces the user to wait for a fix before submitting and relies on the API to report when that happens.

[image]

commit 95d6efd3 (2021-04-13): fix: Always instantiate new ShareDialogFragment

Android doesn't allow new arguments to be passed to an existing fragment after it's been instantiated and attached to a Fragment, so re-using an existing instance doesn't work here because the location won't be updated.

Closes #495, Closes #498
commit 2840a0f0 (2021-11-18): fix: Set the first fix flag with location in addition to GnssStatus

This should fix the "no fix" issue on devices that don't report "on first fix" via the GnssStatus API, because we'll use the Location API callback as well

Closes #495
closed 2021-11-18 by barbeau