barbeau/gpstest #499: Allow contributions via Google Play

barbeau, 2021-04-14:
**Is your feature request related to a problem? Please describe.**
A user expressed an interest to contribute to the apps development via Google Play:
https://groups.google.com/g/gpstest_android/c/9DyyB3MAFns/m/ljQqh-xNBAAJ

**Describe the solution you'd like**
I'll look into this, likely as a separate stand-alone entry on Google Play so the main GPSTest app remains ad-free and in-app purchase free.

From above link:

>Usually, it's to remove ads, or in the case of GPS Essentials, Michael has a "donation plug-in" which works well enough; however, the icon, a penny, is removed once the donation plug-in has been purchased through Google Pay. Ideally, you would present the opportunity to make a donation available anytime, even after having made the first donation, through a single tap from the About screen.

**Describe alternatives you've considered**
https://www.buymeacoffee.com/barbeau exists, but the above user expressed an interest in payments via Google Play

barbeau, 2021-04-22:
Reading:
* https://np.reddit.com/r/Bitcoin/comments/2vro8y/mycelium_and_google_play_recap_of_what_happened/
* https://support.google.com/googleplay/android-developer/answer/9858738?hl=en&ref_topic=9857752 and
* https://pay.google.com/intl/en_in/about/policy/?visit_id=636308113142509927-1253831182&rd=3, which says:

>Fund solicitations
Examples: Solicitation of donations without valid proof of tax exempt status such as, within the United States, a valid 501(c)(3) tax exempt status clearly displayed to the public. Donation solicitations without proof of registration with the relevant country's regulatory bodies and authorities, such as political organizations that have registered with the Federal Election Commission (FEC)

...it seems the safest way to go about allowing contributions without getting banned from Google Play is to publish a new GPSTest app that has an additional trivial cosmetic feature like changing the color palette and charge users for that new app. It's silly, but Google seems to require an exchange of digital goods for the fee. And definitely don't label it "Donate" or "Donation".

barbeau, 2021-04-22:
For anyone reading this, another important point is that Google Play takes 30% of purchases as a transaction fee:
https://support.google.com/paymentscenter/answer/7159343

App price | Transaction fee | Payment from Google to developer
-- | -- | --
$10.00 | $3.00 | $7.00

"Buy me a coffee" (https://www.buymeacoffee.com/barbeau) takes about 10.5% as a transaction fee:

Contribution | Transaction fee | Payment from "Buy me a coffee" to developer
-- | -- | --
$10.00 | $1.05 | $8.95

So if you contribute via sites like "Buy me a coffee", more is going straight to the developer.

vkbellis, 2021-04-22:
Ouch! Neither of those options are good. How about Venmo? I've only used it once, but it went well and the vendor,  https://www.barharboroysterco.com didn't pay for any of the credit card fees. They got their $15 for a dozen of oysters and I was charged the $0.45 service fee on my VISA

barbeau, 2021-04-24:
Alright, I've tried setting up a PayPal.me account here:
https://paypal.me/sjbarbeau/5usd

I've also enabled the Sponsors button on this repo to make this and "Buy me a coffee" options more visible.

vkbellis, 2021-04-24:
This avenue seemed to work on my end https://paypal.me/sjbarbeau/5usd

vkbellis, 2021-04-24:
I'm not seeing the reflection of using PayPal to Buy Me a Coffee
https://www.buymeacoffee.com/barbeau

[image]

barbeau, 2021-04-24:
🤦‍♂️ Thanks, I forgot to connect the two. Should be fixed now.

FYI, the PayPal link should work independent of the Buy Me A Coffee site as a direct payment. Looks like BMAC still has a 3% + $0.30 transaction fee if using PayPal as payment method. PayPal direct payments without going through BMAC shouldn't have any transaction fee.

barbeau, 2021-04-24:
>Hmm, 'button on this repo' - what's a repo?

Sorry, GitHub speak - this repository/project on GitHub. If you view on desktop web browser there should now be a "Sponsor" button with a pink heart up near the "gpstest" project name.

closed 2022-03-14 by stale[bot]