barbeau/gpstest #502: Regarding h and H

vkbellis, 2021-04-20:
From the main menu, Help, About, **Alt** is reported to represent "Meters above the WGS 84 ellipsoid"; however, not all places on the Earth are above the ellipsoid - see image below and Hipsometric Conversation.jpg (attached).

[image]

Consider instead "**Ellipsoid (h) - WGS84 ellipsoid height (m) [EPSG:4326](https://epsg.io/4326)**" noting that https://epsg.io/4326 is only one possibility, and that I do not know what the internal GNSS chip is actually reporting, nor what the OS is actually computing; e.g., transforming Earth-centered-Earth-fixed x, y, z meters into lat, long, ellipsoid meters. Still don't know how or where to find this detail and why the OEMs make finding this information so damn difficult. .

**Alt (MSLS)** is reported to represent "Meters above Mean Sea Level (requires NMEA support)"; however, _MSL_ is kind of vague, and exactly how does NMEA support get used?, does it require the user to do something?, and if not, then why mention it on this screen? Consider instead "**Orthometric (H) - "H ≈ h - N  where H is the orthometric height (m) being approximately equal to the ellipsoid height h (m) - the geoid height (m).**" The specific geoid model used by GPSTest needs to be displayed with a link provided to its expanded explanation.

barbeau, 2021-04-21:
@vkbellis Thanks for opening this!

>Alt is reported to represent "Meters above the WGS 84 ellipsoid";

This comes directly from the text on the `Location.getAltitude()` method where GPSTest gets the altitude information from Android:
https://developer.android.com/reference/android/location/Location#getAltitude()

I'd prefer to match the official Android docs rather than changing the definition, because ultimately that's all I can rely on.

>Alt (MSLS) is reported to represent "Meters above Mean Sea Level (requires NMEA support)";

I explained this a bit more in https://github.com/barbeau/gpstest/issues/503#issuecomment-824199949. In short, this value gets parsed from NMEA sentences from other Android APIs, which may not be supported on all devices, hence the caveat. 

This also creates some problems with the documentation itself, because the official NMEA spec isn't public, so I'm getting a data stream that isn't well documented. The existing text comes from the reference I linked to here - https://github.com/barbeau/gpstest/issues/22#issuecomment-260234637. That being said, if anyone has a better reference for this information and official definitions I'm open to improving them.

vkbellis, 2021-04-22:
RE: **Alt**, in geodesist circles, the word 'up' is used somewhat conveniently, but Ellipsoid height (h) is more clear, in my opinion, if we're referring to heights relative to the ellipsoid.

RE: **Alt (MSL)**, since it creates problems, is giving the wrong values, and is so poorly documented, consider not showing it at all. In its place, consider a link or 2 to available geodetic tools like VDatum and VERTCON (excellent desktop programs). The Android app by RDH Software, "My Elevation" is pretty good (I've done some testing using it against actual survey data). It uses its own sort of "fused" approach by combining data from various sources. But it contains ads and Mike wrote this when I asked if they might be removed: "_There isn't a paid version.  If there is one in the future it's likely to be a subscription rather than a one time payment. The whole industry is finding one time payments don't really work for apps that have their own use-based costs in the background (monthly fees for all the services the app needs to function).  In addition to a subscription removing ads, I'd want to add some additional, significant, features to be included in that but the truth is my day job is really keeping me from getting to it.  So for the foreseeable future it will be just minor new features and updates to keep current with the latest Android changes. 

Mike
RDH Software_"

gdt, 2021-06-03:
I find "meters above WGS84 ellipsoid" to be both clear (negative is therefore below so that's fine) and aligned with the Android spec.  There are dragons lurking: the ellipsoid needs to be anchored in space for that to be meaningful, so we're talking datum not just ellipsoid, and there are a bunch of realizations of WGS84 -- but obviously we mean the one in use at the moment.   Further dragons are that chips compute positions in the frame of WAAS or whatever correction source is in use, and do something undocumented when combining pseudoranges from muliple contstellations.  However, WGS84(G1762) and ITRF2008 are essentially the same, as are all the other modern frames.

As for "WGS84 Orthometric Height" aka "AMSL", that is well defined as the distance above the WGS84 reference equipotential surface, and computed from ellipsoidal height and the WGS84-specified geoid model EGM2008.   The dragon here is that the GPS chips have really far off approximations (4m error in my area) for this geoid model, for no apparent reason.  But the right answer is well defined.    The help text should probably say that the AMSL height is the orthometric height reported by the receiver.

The NMEA spec is quite available; it's just copyrighted so it's not available on the Internet.    My impression is that NMEA predates GPS and thus the basic concept is "height above the sea level reference surface" and that it was from a time before a few meters from geoid models was discernable.   I found a trimble page which talks about sea level reference but does not talk about datum and geoid:
  https://www.trimble.com/OEM_ReceiverHelp/V4.44/en/NMEA-0183messages_GGA.html

barbeau, 2021-09-23:
I'm updating the "Help" text to the following to address the above raised issues:

><b>Alt</b> - Meters above the WGS 84 ellipsoid, also known as h
        <b>Alt (MSL)</b> - Meters above Mean Sea Level, also known as Orthometric height or H. H = h - N, where N is the height of the geoid above the WGS84 ellipsoid. Parsed from internal NMEA stream, so it is only available if your device supports this API. Because this data comes from the chipset, the exact geoid model is unknown, however based on https://issuetracker.google.com/issues/191674805#comment16 it seems Qualcomm devices are using EGM84.

commit d76112fe (2021-09-23): docs: Update Help definitions for "Alt" values

Closes #502
closed 2021-09-23 by barbeau