barbeau/gpstest #505: Logs & Logging

vkbellis, 2021-04-22:
**Is your feature request related to a problem? Please describe.**
Yes, in part, some of the following suggestions are related to issues.

**Dedicated Folders and more clearly named logs**
GPSTest and GnssLogger appear to want to share a common folder on my phone by default:
../storage/emulated/0/gnss_log/ 

This becomes even more confusing trying to discern which app created which file by their similarities in file naming schema, for example, here we see both apps' files with (2) from GPSTest:

[image]

Consider having GPSTest by default create its own parent folder; e.g., ../storage/emulated/0/GPSTest_logs/ 
and then beneath it, sub-folders for each instance that GPSTest in run; e.g., 
../storage/emulated/0/GPSTest_logs/202104180600/
../storage/emulated/0/GPSTest_logs/202104180300/
../storage/emulated/0/GPSTest_logs/202104151530/

and then within each sub folder, clearly named log files; e.g.,
gnss_log_nmea_2021_04_19_13_42_52.nmea
gnss_log_nav_messages_2021_04_19_13_42_52.21N (.yyN)
gnss_log_rinex_2021_04_19_13_42_52.21o 
gnss_log_location_fixes_2021_04_19_13_42_52.txt
gnss_log_antenna_info_2021_04_19_13_42_52.csv
gnss_log_antenna_info_2021_04_19_13_42_52.json

**Six Checked Boxes - Only Two Logs**
Next, from the main menu, Settings, scrolling down to the last item, Logging and Output, we see under the File Output heading that (6) logs should be produced when the boxes all have been checked, but in my tests thus far (GPSTest v3.9.11), only (2) files are being written. 

**Separate Settings for Logging, Android Studio**
Under the heading Android Monitor Output, we see (4) boxes, and that when these boxes are all checked, it's unclear what, if anything, is going.

[image]

Android Studio settings (keep under main menu, Settings) ought to be separate and apart from Logging Settings.

**Toggle Logging On/Off**
Consider inserting the word Logging between Status and Map under the main menu with a slider toggle switch to turn logging on and off. Presently, it's not at all clear when any logging is begun or ended.

**Saving & Sharing Logs**
Under Logging and Output, consider adding options for users to
define their default log directory
define their email recipient(s)
define their remote storage locations

From the Status screen, sharing icon, Log, consider adding the option for the user to zip all files under the active child folder (or make this the default behavior with option to not zip), before the act of sharing. There might also be a cautionary pop up alert when the user has not previously and overtly stopped logging, and that by going ahead with sharing, all active logging will be halted and saved automatically before sharing.

barbeau, 2021-04-22:
@vkbellis Thanks for opening this! Could you please split into individual issues with one topic per issue? It makes it easier to address/discuss items.

barbeau, 2021-04-23:
@vkbellis Thanks for splitting this up! I'm going to close this one as I believe all the above topics are in separate issues now.

closed 2021-04-23 by barbeau