barbeau/gpstest #508: Six Checked Boxes - Only Two Logs

vkbellis, 2021-04-22:
RE: GPSTest v3.9.11

From the main menu, Settings, scrolling down to the last item, Logging and Output, we see under the File Output heading that (6) logs should be produced when the boxes all have been checked, but in my tests thus far (GPSTest v3.9.11), only (2) files are being written. 

[image]

barbeau, 2021-04-23:
So this is actually expected behavior:
https://github.com/barbeau/gpstest/blob/master/LOGGING.md#logging-data-to-files

All "CSV" output is logged to the same file, and all JSON output is logged to the same file.

I'll leave this issue open and see if I can change the in-app text to make this behavior clearer (maybe "the file" instead of "a file").

vkbellis, 2021-04-23:
Having one big ole file with observations and nav messages is possibly the cause of not being able to post-process these smartphone files in any survey program. Why must they all be in one common bag? Is there any app or means to neatly extract the multiple files from the single mega file?

barbeau, 2021-04-23:
>Why must they all be in one common bag? Is there any app or means to neatly extract the multiple files from the single mega file?

This is Google's format that they created with GnssLogger. It's compatible with Google's desktop analysis tool based on Matlab. See:
* https://developer.android.com/guide/topics/sensors/gnss#log
* https://github.com/google/gps-measurement-tools

GPSTest follows this same format so it's compatible with the same tool.

barbeau, 2021-04-23:
If you want to break the records up yourself, you basically just need to process them in any program that accepts CSV, group them by the first column values, and then trim off the first column. You could do this manually in a spreadsheet program like Excel.

vkbellis, 2021-04-23:
OK. Thank you Sean.

commit 12c98316 (2021-04-28): docs: Clarify that logs are output to single file

Closes #508
closed 2021-04-28 by barbeau