barbeau/gpstest #514: Needed for Post-processing: Navigation Messages

vkbellis, 2021-04-23:
In order to post-process RINEX files, it would be wonderful if GPSTest could log navigation messages.

REF:
https://developer.android.com/reference/android/location/GnssNavigationMessage
https://developer.android.com/reference/android/location/GnssNavigationMessage.Callback

barbeau, 2021-04-23:
Good news - it does! See https://github.com/barbeau/gpstest/blob/master/LOGGING.md#data-output---csv

The issue is that most devices don't support this Android API - see Column O "Navigation messages" in the [GPSTest Database](https://bit.ly/gpstest-device-database).

closed 2021-04-23 by barbeau

vkbellis, 2021-04-23:
There are numerous survey applications which require this navigation information in order for any post-processing to happen.