barbeau/gpstest #515: L2C and E5b are shown as L1 and E1

wdc-rsat, 2021-04-23:
**Describe the bug**
GPSTest shows the carrier frequency (CF) indicator for L2C as L1 and for E5b as E1, although their numeric values provided within the raw measurement data are correct (1227.6 and 1207.14) 

**Device and Android version:** 
Xiaomi Mi8 Pro (equuleus), LineageOS 17.1 (unofficial build)

barbeau, 2021-04-23:
Hmmm, strange. I didn't think the Mi 8 Pro supported L2C and E5b? Could you share the raw measurements data here?

My guess here is that the raw measurement data actually has bad carrier frequency values. This isn't uncommon with unofficial ROMs.

The in-app "Status" display of carrier frequency labels is driven by the Android `GnssStatus` API:
https://developer.android.com/reference/android/location/GnssStatus

...while the raw measurements log comes from the `GnssMeasurements` API:
https://developer.android.com/reference/android/location/GnssMeasurementsEvent

The mappings of frequencies to labels is here, and you can see that L2 and E5b labels should be shown for those frequencies if they were in the `GnssStatus` data:
https://github.com/barbeau/gpstest/blob/master/GPSTest/src/main/java/com/android/gpstest/util/CarrierFreqUtils.java#L46

So my guess is that the Mi 8 Pro is outputting the correct CF values from `GnssStatus`, and bad ones from `GnssMeasurements`.

wdc-rsat, 2021-04-25:
> I didn't think the Mi 8 Pro supported L2C and E5b?
I have followed your advice and edited the /vendor/etc/gpsconfig.xml adding
MultiCarrRFMode ="GL_MULTI_CARR_RF_MODE_L1_L2_L5"

> the raw measurement data actually has bad carrier frequency values.
GnssMeasurements come directly from the Broadcom /vendor/bin/glgps binary
and GnssStatus API is a thin android java wrapper written by the handset manufacturer.
I have more trust in Broadcom's GNSS competence and modified the rokubun/android_rinex 
converter script to support L2 and E5b, but it is better to have an independent verification.

barbeau, 2021-04-27:
This doesn't seem like an issue in GPSTest, so I'm going to close this. Please feel free to comment with additional info if the problem isn't solved.

closed 2021-04-27 by barbeau