barbeau/gpstest #516: Missing Nav info in logs

vkbellis, 2021-04-23:
GPSTest v3.9.11

In 108,362 lines of data in the log file, only 1 line even mentions Nav:

[image]

[gnss_log_2021_04_23_12_05_40.zip](https://github.com/barbeau/gpstest/files/6368052/gnss_log_2021_04_23_12_05_40.zip)

barbeau, 2021-04-23:
@vkbellis See my comment in https://github.com/barbeau/gpstest/issues/514#issuecomment-825744442. Most devices don't support this, so you won't see it unless you have a supported device.

closed 2021-04-23 by barbeau

vkbellis, 2021-04-23:
There are numerous survey applications which require this navigation information in order for any post-processing to happen.

barbeau, 2021-04-23:
>There are numerous survey applications which require this navigation information in order for any post-processing to happen.

Yup. You can this data from other sources, though, so it shouldn't be a deal-breaker.

barbeau, 2021-04-23:
Just to be clear, I don't have any control over devices not supporting navigation messages in GPSTest - you'd need to go talk to the OEMs for that :).

vkbellis, 2021-04-23:
I see now that my phone (Samsung Galaxy A71 G5 is one of the devices that don't support Nav messages - thanks again for that great database!

Guess I'll try my luck at getting the nav files from https://cddis.nasa.gov/ next.

Thank you again for pointing out that the issue is related to the device's lack of support.