barbeau/gpstest #519: README playstore link is broken?

OmarShehata, 2021-05-28:
Clicking on the README link to go to the playstore takes me to: https://play.google.com/store/apps/details?id=com.android.gpstest

Which says:

```
We're sorry, the requested URL was not found on this server.
```

[image]

I was going to open a PR and update the link but I can't seem to find the correct app on the Playstore anymore? Although I had downloaded it on my Android phone a while ago.

barbeau, 2021-05-29:
@OmarShehata thanks for opening this! Google has removed the app from Google Play for some currently unknown reason related to location permissions, and I've filled out the appeal form and am waiting to hear back. I'll post here and in the Google Group when I get more info and hopefully GPSTest is restored on Google Play.

In the meantime you can still download the APK from the releases section here on GitHub or Fdroid as linked in the README.

OmarShehata, 2021-05-31:
Great, thanks for confirming! Eagerly awaiting to see this amazing app back on the store.

barbeau, 2021-06-02:
And we're back! GPSTest is live on Google Play again. 

No response from Google yet on exactly why it was initially removed. I did notice that in the "Policy" section the "Uses background location" box was checked, which I don't recall checking. I unchecked it and re-submitted it for review. I'm wondering if Google flipped this themselves when they incorrectly flagged a past update as using background location (long story - see https://github.com/barbeau/gpstest/issues/456 for details).

closed 2021-06-02 by barbeau