barbeau/gpstest #523: Reveal Datums From NMEA Sentences

vkbellis, 2021-06-10:
**Is your feature request related to a problem? Please describe.**
To be clear, there's not a problem with GPSTest. The problem in general is a lack of transparency from OEMs regarding the reference frame and gravity model that they're using. NMEA sentences begin to tease out that information and what remains to be seen is how these data vary between OEMs, Platforms, and beyond irrespective of OS.

**Describe the solution you'd like**
There are various NMEA sentences that contain indications, but the three primary sentences types are GGA, DTM and GNS
[image]
Other NMEA sentence types containing similar frame and model information relate to RTK and their applicable base stations; however, the first three listed here would probably be adequate for research purposes. Because GNS and GGA lines also contain specific positional data, the user sharing their GPSTest results to the crowd-sourced database may prefer to have their positional data redacted, and that's fine.

**Describe alternatives you've considered**
At this time, I know of no available tools that help in identifying the device's intrinsic frame and geoid model.

**Additional context**
While much of this suggested enhancement for GPSTest stems from a need to get to the origin of the widely reported and inexplicable 20+ meters of error in ortho heights, uncloaked knowledge of the datums used in solutions should be fundamental.

barbeau, 2021-06-16:
@vkbellis Thanks for opening this.

So given a DTM NMEA sentence:

```
$GNDTM,P90,,0.000021,S,0.000002,E,0.989,W84*57,,,,,,,,,,,,
```

You'd like to see two new columns in the Database, like:

| NMEA DTM Local datum code | NMEA DTM datum
| --- | --- |
| P90 | W84 |

Is that right?

vkbellis, 2021-06-18:
@barbeau - I'm thinking you may not appreciate me opening this ticket, and I totally understand if you don't want to implement any of this, really, it's okay if you don't. But this at least allows for an exchange of ideas, so in answer to your question, this is more like what I was asking of GPSTest:

[image]

There are competing thoughts which need some explaining. Beginning with location, the NMEA sentences GGA and GNS both contain Lat/Lon values. These geographic positions are necessary in two separate calculations independently determining 1) the predicted gravity value (i.e., geoidal separation (N)) at the given location and 2) the predicted orthometric height (H). The device's reported ellipsoidal height (h) is also needed for evaluating the veracity of the smartphone's GNSS chops. The counter point of privacy defeats the purpose and so the idea emerges: dumb down the Lat/Lon *display* in the Google Sheet by rounding to the nearest minute, but leave the data fully intact in the downloaded files And lastly, allow users when sharing their data to opt out from sharing positional information details.

barbeau, 2021-06-19:
>I'm thinking you may not appreciate me opening this ticket, and I totally understand if you don't want to implement any of this

Not at all! I really do appreciate the time you've taken, especially in #503, to explain and help investigate what's going on. I appreciate your deep knowledge of surveying and GIS, which is better than mine 😁. I've wanted to dive deeper into the h issues for a while but just haven't had the bandwidth. Please don't interpret short answers as being uninterested or annoyed, this is a side project for me that I usually work on on my lunch breaks so sometimes the choice is short answers or no answers.

Let me think about this one further. I really want to implement something here I just need too figure out exactly what, balancing user privacy and also needing to limit this to a single row in the spreadsheet due to implementation issues.

For the contents of the DTM string itself, how precise and unique is the gravitational model used too determine offsets? Knowing this and lat/long rounded to the minute, could you reverse engineer a more precise lat/long of the user? What about knowing offsets and country?

ValZapod, 2022-01-25:
> I know of no available tools that help in identifying the device's intrinsic frame and geoid model.

Really?? Not even internally to google?

barbeau, 2023-09-20:
Not stale

barbeau, 2024-04-11:
Not stale

barbeau, 2025-05-16:
Not stale