barbeau/gpstest #527: Parse DTM and display some info if it is non-standard

gdt, 2021-06-21:
We have observed receivers that operate in strange datums, when the user has not configured them to do so (.e.g. PZ-90 in the US).   This is a receiver bug, not a GPSTest bug, and thus this is a feature request for GPSTest to help the user understand what their receiver is doing.

While a user can log data and go read NMEA, that requires effort.  It would be great if GPSTest would observe DTM sentences and parse them, and if a non-standard configuration is observed, display something to the user.  By non-standard, I mean anything other than "just WGS84, period, no offsets".   What could be displayed might just be DTM in red, to give a clue that the raw NMEA should be looked at, or it could be more, like "DTM: PZ-90".    (That's P90->PE90 from NMEA, and PZ-90 because that's what EPSG calls it.)

Perhaps, the red DTM could be clickable, and go to a screen with the full contents including transforming the minutes E/N shift into cm.    But really, just seeing a '"your receiver is operating in a non-standard datum" warning would be a huge step forward.

barbeau, 2021-06-22:
Thanks @gdt, this is a good idea. I've added to the future "Device info" screen that I have planned in https://github.com/barbeau/gpstest/issues/313#issuecomment-513276069, which will also show other common data issues. I'm going to close this issue as a duplicate now that I have this captured in https://github.com/barbeau/gpstest/issues/313#issuecomment-513276069.

closed 2021-06-22 by barbeau

Bug725, 2021-06-23:
unsubscribe