barbeau/gpstest #531: Add mapsforge, or really stop using proprietary maps APIs

gdt, 2021-06-21:
The f-droid build doesn't seem to have working maps.   Presumably this is because the app uses proprietary APIs which are avoided.

Add mapsforge, so that users can display maps without proprietary OS components, and without a privacy violation.  Look in the mapsforge directory of the sd card for maps, so they can be shared with others.   See OpenTracks OSM provider and SatStat for apps that do this.

barbeau, 2021-06-22:
@gdt Yes, allowing users to load their own map tiles for the OSMDroid variant is a good idea. It currently uses MapTiler free tier, which tends to get exhausted - see https://github.com/barbeau/gpstest/issues/316. So if you try near the beginning of the month it should work, but will eventually stop working when the quota is reached.

gdt, 2021-06-22:
Thanks.   MapTiler's privacy policy looks good, but still it would be nice if the README.md explained that the users location is implicitly sent to third-party servers when clicking the map tab.

cewbdex, 2021-06-23:
I'm not sure if it is really possible, but I would say a lot of fdrodid users have OSMAnd installed, as it's probably the best foss navigation app, maybe it would be possible to reuse its offline maps too?

gdt, 2021-06-23:
Good point about osmand, but that has come up for discussion in many other apps that want offline map capability.   The short story is that the format is somewhat complicated and more importantly there is no library that will render it that can be included.  I specifically mentioned mapsforge because there is an embeddable library to do what is needed, which is to have some app that is only wanting a map incidentally or as a background layer to display it easily. 

Yes, there is data duplication, but in my case I have OsmAnd maps for a lot of places (because if I am there navigation is very important) and I have a mapsforge map for  smaller area, currently the 6 states of the US northeast, 620 MiB.

So if you can do the work in the OsmAnd world to have a library that is usable within apps that reads the obf files, and does more or less what mapsforge does, definitely lots of people would be very interested in considering it.

barbeau, 2021-06-25:
>but still it would be nice if the README.md explained that the users location is implicitly sent to third-party servers when clicking the map tab.

@gdt Good point, I updated the GPSTest privacy policy to reflect this - https://github.com/barbeau/gpstest/wiki/Privacy-Policy.

AntoninCurtit, 2022-03-01:
Hi,
Thanks for using OpenStreetMap !
Please, don't forget to credit it by adding "© OpenStreetMap contributors" as required by the license (osm.org/copyright)

Thanks !

barbeau, 2023-10-24:
Not stale

gdt, 2024-05-11:
@stale Bot considered harmful!  Written in Beautiful Handwriting even.

barbeau, 2025-02-25:
A source for tiles - https://www.openandromaps.org/downloads/laender-und-regionen