barbeau/gpstest #532: Point Averaging

vkbellis, 2021-06-23:
**Is your feature request related to a problem? Please describe.**
Nope.

**Describe the solution you'd like**
Instead of seeing individual fixes as they happen, display the mean and 3 sigma.

**Describe alternatives you've considered**
None.

**Additional context**
Add any other context or screenshots about the feature request here.

barbeau, 2021-06-23:
@vkbellis Thanks for the suggestion! To clarify:
1. You'd like to see this on the Status screen?
2. You'd want lat, long, and alt averaged and shown separately?
3. You'd want it averaged over the entire session (versus a sliding window of a certain number o points)?

Also note there is an issue open here for showing an averaged position on the Accuracy view - https://github.com/barbeau/gpstest/issues/374

Related to this - in your opinion what's the "right" way to average a position given lat and lon, or lat, long, and HAE? Does this sound right - https://math.stackexchange.com/a/47856/554287?

vkbellis, 2021-06-25:
@barbeau  Thank you for considering my suggestion.

1. The Status_screen is one possibility, though there are other possibilities depending on how you want to handle displaying the statistical data including adding a new display screen before or after the Accuracy_screen. I think keeping averaging and accuracy as two separate ideas on separate display screens might be better because the actual published/ known position is fundamental to the latter and not requisite for the former.
2. Yes, mean stats for Lat/Lon **HRMS** and _ellipsoid heights (h)_ **VRMS** and possibly 3dRMS.
3. A tap of the Start/Stop Averaging button would be nice; adding an optional 00:00:00 individual rolling setting timer would be excellent!

The display ought to make clear the height of device, antenna reference point vertical height (ARP HI), and that it's been accounted. Related; name the occupied station, applicable to both Accuracy_screen and Averaging_screen.

As for the mathematical approach, converting the API's lat/long/h to x,y,z, mean, then back to lat/lon/h is also mentioned in this posting which contains some further discussion: https://gis.stackexchange.com/questions/7555/computing-an-averaged-latitude-and-longitude-coordinates

fgnievinski, 2022-04-04:
this is commonly discussed request, e.g.:
- https://forums.geocaching.com/GC/index.php?/topic/352891-gps-averaging-app-for-android/
- https://www.reddit.com/r/geocaching/comments/mny9iu/android_gps_averaging_coordinate/

it's the subject of another open source app, which is no longer available in play store -- maybe its code could be incorporated here?
https://github.com/davidvavra/GPS-Averaging

the feature is available from other closed-source free apps, for example:
- https://play.google.com/store/apps/details?id=com.jcl.GPSAverage&hl=en&gl=US
- https://play.google.com/store/apps/details?id=precise.GPS.v2.free&hl=en_US&gl=US
- https://play.google.com/store/apps/details?id=com.sciencewithandroid.precisiongpsfree
- https://play.google.com/store/apps/details?id=menion.android.locus

barbeau, 2023-11-27:
Not stale

k0styan3, 2024-03-26:
👍