barbeau/gpstest #533: Add reset for accuracy screen

gdt, 2021-06-24:
I was standing someplace, started GPSTest, ran it for a while to watch the satellite count go up, and then put it on a (nearby but not exactly) mark and started accuracy.   It seemed to have samples from before, and there was no obvious way to clear them to only average the accuracy from after my desired start time.

I'd like two things.  One is a reset button that after it is pushed, data from before the push is not included in the stats.
The second thing is that when accuracy mode is entered, an implicit reset should happen.  There's no reason to believe that the user was in position before invoking accuracy mode, and I at least don't expect the past to be included.

vkbellis, 2021-06-25:
The accuracy counter will automatically reset to zero by simply tapping on the edit button without actually editing your known location's value.

[image]

barbeau, 2021-06-25:
@gdt Yes, for now a workaround is just to tap "Edit" and then "Save" again for your ground truth location, and it will reset the counter back to 0. I agree more explicit controls would be useful here, and I'll leave this issue open for future improvements.

gdt, 2022-08-14:
Hey @stale  this is still valid.

gdt, 2023-03-03:
Hey https://github.com/Stale this is still valid.

barbeau, 2025-05-17:
Not stale