barbeau/gpstest #535: Seek Help for using Glonass for fix

jlmxyz, 2021-06-27:
**Is your feature request related to a problem? Please describe.**

it's not related in particular with gpstest, but you seems to be well aware of Android GNSS internals.... and currently I only found information that are useless, I even tried to look into AOSP code, but since there is really few information about how android fix is computed, I only found some differences in gps/msm8084/utils/loc_nmea.cpp but nothing that seems to be related to glonass....

I've a n910f that show glonass and beidou satellites under gpstest, they all have azimuth and elevation information and some have CNO information.
Log time ago (several years, using another test app) I used to be able to use Glonass satellites for fix in addition of gps one
I found your article https://barbeau.medium.com/dual-frequency-gnss-on-android-devices-152b8826e1c that since 7.0 there was a support for multiple constellations, it was about this time that I lost the Glonass in fix....
I've a custom rom currently running Android 10 and Android 11

**Describe the solution you'd like**
if you have any information about gps.xml modification that could re-activate use of glonass information in fix, or any information were to look in AOSP sources, it would be marvelous!!!

thanks in advance
JLM

barbeau, 2021-06-28:
Yes, the driver interface for Android 7 and up changed because of the introduction of official support for multiple GNSS constellations - previously all GLONASS satellites had been hacked into GpsStatus objects with PRNs within a certain range. So unofficial ROMs that don't have direct OEM driver support are likely to break if you're trying to bring a pre-7 device to post-7.

You could try unchecking the GPSTest setting option for "Use GNSS APIs", which would tell GPSTest to listen to the older pre-7 API for updates, which might show GLONASS being used, but then I'm guessing you'll lose Beidou (?). Seeing before and after screenshots for this setting would help.

Because this isn't strictly a GPSTest issue I'm going to close this out, but please feel free to comment more if you find something that works.

closed 2021-06-28 by barbeau

jlmxyz, 2021-07-02:
here is a screenshot with GNSS api unchecked

jlmxyz, 2021-07-02:
and a few seconds later, with GNSS API checked
[image]

jlmxyz, 2021-07-02:
there is no difference, fix time can be explained by having a previous fix and same with accuracy.
do you have a reference document of what is required by a vendor to be compliant with GNSS api?  or a document explaining how GNSS api works?

barbeau, 2021-07-02:
There is this AOSP documentation generally:
https://source.android.com/compatibility

...and you can find some of the C++ header and code information in AOSP here:
https://android.googlesource.com/platform/hardware/libhardware/+/master/include/hardware/gps.h

But that's about i that I'm aware of, and I'm not sure that's helpful in your case. I'm assuming something is broken in the GNSS binary drivers that only the OEM could fix.

There is also the possibility that your device is in fact using the other constellations to compute the fix but the Android GnssStatus API just isn't getting the status info for the flags like "used in fix" from the hardware, which is what GPSTest shows in the UI. IIRC a Motorola device had that bug - I think there is a closed issue on that.