barbeau/gpstest #538: Is it a misjudgment that the Beidou signal of the Mi 8 in the GPSTest Database is marked as B1C instead of B1I?

Jia-le-wang, 2021-07-03:
I noticed that the Beidou signal of the Mi 8 phone is marked as B1C signal, whose frequency is 1575420030Hz.However, from the data recorded by GNSSlogger, the Beidou signal captured by Mi 8 is still B1I, whose frequency is 1561097980Hz.

In brief, the Beidou B1C signal and B2a signal are new signals of the Beidou 3 system, which are transmitted from Beidou's C19-C46 satellite.In fact, only a few recent phones support these two new signals, such as the Mi 11, Huawei P40, Huawei Mate40, etc.

The attached file is a small piece of data collected with GNSSlogger. The constellation type 5 corresponds to the Beidou signal of MI8.

[image]
[MI8-gnsslogger.txt](https://github.com/barbeau/gpstest/files/6758441/MI8-gnsslogger.txt)

barbeau, 2021-07-03:
@Jia-le-wang Thanks for opening this!

It looks like this is indeed a bug in the Mi 8. The display of carrier frequency bands in GPSTest (both the Status view and Database) is driven by data from the GnssStatus API:
https://developer.android.com/reference/android/location/GnssStatus

The log from GnssLogger and GPSTest uses the GNSS raw meaurements API:
https://developer.android.com/reference/android/location/GnssMeasurementsEvent

It looks like the GnssStatus API is erroneously reporting 1575420030Hz as the carrier frequency, while the raw measurements API is reporting 1561097980Hz as the carrier frequency. They don't agree with one another, so it appears the GnssStatus API is outputting bad data.

From a quick glance at the database it appears other devices with the Broadcom BCM4775 chips are reporting the same, so I wonder if this is actually a Broadcom bug.

Could you please report this to Xiaomi and let me know what they say?

Jia-le-wang, 2021-07-04:
OK，barbeau！ I will try to give feedback on this issue, but I am not sure if Xiaomi will respond to me.Once I know the reason, I will reply here as soon as possible.

barbeau, 2022-01-25:
Closing this for cleanup as it's not a GPSTest app issue, but I'd be interested in knowing if Xiaomi fixes (fixed?) this.

closed 2022-01-25 by barbeau