barbeau/gpstest #543: Accessibility - add text labels to buttons

HegrJan, 2021-08-15:
Please add text labels to the buttons in Share dialog (or use another accessibility feature to describe them to the assistive apps). Their meaning is indicated just by an icon which is meaningless for the blind users.
The screen-reader presents all the buttons as "button" without any hint what could it do.

barbeau, 2021-08-16:
@HegrJan Thanks for pointing this out! I'll work on fixing it.

barbeau, 2021-08-19:
@HegrJan Ok, I've gone through the app and added content descriptions to the various buttons and ImagViews used throughout. 

I'll try to roll out a beta update to the app soon. You can enroll in beta updates at https://play.google.com/apps/testing/com.android.gpstest. I'd love to get your feedback on the next beta release to see if it's a better experience and what else could be improved.

closed 2021-08-19 by barbeau

barbeau, 2021-08-20:
@HegrJan I'm rolling out v3.9.14 to beta testing on Google Play now if you could give it a shot! Should be available within 12 hours or so. F-Droid version is available here to download if you'd prefer that:
https://github.com/barbeau/gpstest/releases/tag/v3.9.14

commit 6004845d (2021-08-19): fix: Improve app accessibility

Adds content descriptions to ImageViews and buttons used throughout the app

Closes #543