barbeau/gpstest #545: Various GNSS CFs data

KovalevArtem, 2021-08-16:
**Describe the bug**
The test results of the application v3.9.12 differ from the version v3.9.13

**To Reproduce**
Steps to reproduce the behavior:

- All measurements were taken at the same location.
- The measurements were taken with a difference of a couple of minutes.
- At first, I installed googleRelease-v3.9.12.apk then completely uninstall and install googleRelease-v3.9.13.apk (currently in beta)
- Naturally, the Android version was the same on each test ...

**App, Device and Android version:** 
Sony H8266 - Android 10
Xiaomi M2002J9G - Android 11
Sony J9110 - Android 10
Samsung	SM-A217F - Android 11

[googleRelease-v3.9.12.apk](https://github.com/barbeau/gpstest/releases/tag/v3.9.12)  and [googleRelease-v3.9.13.apk ](https://github.com/barbeau/gpstest/releases/tag/v3.9.13)

**Screenshots:** 
[image]

Issue was created based on [this post](https://github.com/barbeau/gpstest/issues/520#issuecomment-899489671)

barbeau, 2021-08-18:
@KovalevArtem Thanks for the report!

A few questions:
1. How long did you let the app run before submitting the data to the database?
2. Could you please post Status screenshots from a before and after for at least one of these devices? I'd like to see if the values in Status match the values shown in the chart.
3. I'd also be curious to know if you installed, uninstalled, and re-installed the same version if you see differences.

IIRC the only changes in GPSTest that would result in different values in the spreadsheet are for Android 12 devices, and that would only be for if raw measurements, carrier phase, navigation messages, and GnssAntennaInfo are supported.

As I mention in https://barbeau.medium.com/crowdsourcing-gnss-capabilities-of-android-devices-d4228645cf25, GPSTest can only observe real-time conditions as reported by the GNSS chipset for supported CFs and GNSSs and these can be transient on some devices. You're most likely to get consistent results if you wait for a while (e.g., 5 minutes, outdoors with clear sky) before reporting capabilities.

KovalevArtem, 2021-08-28:
<details>
<summary>Sony H8266 v3.9.12</summary>

[image]

</details>

<details>
<summary>Sony H8266 v3.9.13</summary>

[image]

</details>
------------------
<details>
<summary>Xiaomi M2002J9G v3.9.12</summary>

[image]

</details>

<details>
<summary>Xiaomi M2002J9G v3.9.13</summary>

[image]

</details>
------------------

<details>
<summary>Sony J9110 v3.9.12</summary>

[image]

</details>

<details>
<summary>Sony J9110 v3.9.13</summary>

[image]

</details>

KovalevArtem, 2021-08-30:
Re-tested and compared application versions v3.9.12 to v3.9.14

- All measurements were taken at the same location.
- The measurements were taken with a difference of a couple of minutes.
- At first, I installed googleRelease-v3.9.12.apk then completely uninstall and install googleRelease-v3.9.14.apk (currently in beta)
- Naturally, the Android version was the same on each test ...

**App, Device and Android version:** 
Sony H8266 - Android 10
Xiaomi M2002J9G - Android 11
Sony J9110 - Android 10
Samsung	SM-A217F - Android 11

[googleRelease-v3.9.12.apk](https://github.com/barbeau/gpstest/releases/tag/v3.9.12)  and [googleRelease-v3.9.14.apk ](https://github.com/barbeau/gpstest/releases/tag/v3.9.14)

[image]

The results are the same!

barbeau, 2021-08-30:
@KovalevArtem That's good! So it seems the differences in the first groups of test may have been due to the amount of time GNSS had been active on each device before submitting results (with less time active = fewer CFs and GNSSs). Unfortunately this level of variability is expected and why I'm leaving multiple records per model in the database so users can see different results from different executions.

I'm going to close this issue as it's not a problem with GPSTest per se, although I'm opening https://github.com/barbeau/gpstest/issues/546 to submit time-to-first-fix and/or GNSS uptime to hopefully reflect these metrics in the data.

closed 2021-08-30 by barbeau