barbeau/gpstest #547: Add screenshots

aha999, 2021-09-04:
**Is your feature request related to a problem? Please describe.**
Currently we can't see what the app looks like prior to installing.

**Describe the solution you'd like**
Add screenshots to github readme and F-Droid repo.

barbeau, 2021-09-21:
Thanks for the feedback! Indeed, this is on my TODO list, and I actually have another issue open here for it:
https://github.com/barbeau/gpstest/issues/470

So I'm going to close this one.

barbeau, 2021-09-25:
Reopening, as the last release didn't seem to change anything - I'm not sure why:
https://f-droid.org/packages/com.android.gpstest.osmdroid/

barbeau, 2021-09-25:
Whoops, wrong issue - I'm going to reopen the other

closed 2021-09-25 by barbeau