barbeau/gpstest #549: App showing what api the device+ROM is supporting

BurhanBudak, 2021-09-23:
**Check if the phone and ROM implemented the location based API**
If it possible to add a tab in the app showing boolean value if a API is present in the phone. The OEM or ROM devs doesnt always add support.  There could also be false postivies where the phone interprets the unsupported signel but shows it as something familiar.

barbeau, 2021-09-23:
@BurhanBudak thanks for the feedback! I am planning to do this as part of https://github.com/barbeau/gpstest/issues/313, so I'm going to close this issue as a duplicate. Feel free to leave comments in that other issue if you'd like.

closed 2021-09-23 by barbeau

BurhanBudak, 2021-09-23:
> 
> 
> @BurhanBudak thanks for the feedback! I am planning to do this as part of #313, so I'm going to close this issue as a duplicate. Feel free to leave comments in that other issue if you'd like.

thx for the update. will check out #313