barbeau/gpstest #553: Un-cap data precision for lat/long/height when logging to CSV, and expose height accuracy

tarquinwj, 2021-10-04:
Currently, when logging to csv, the lat/long/height data is capped to 6 decimal places. This means that lat/long is capped at about 10 cm precision, and height is capped at 0.001 mm. (By contrast, GNSS Logger uses 7 dp for lat/long and 6 dp for height.)

For the purposes of averaging (or performing statistical analysis of) the data, it would be much more useful to log this at the precision level that it comes from the GNSS chip, which is 14 decimal places.

It would also be useful to expose the height accuracy (EPE) in the log. Currently only the horizontal accuracy is logged. Having access to both would make it possible to perform weighted averaging or weighted statistical analysis. GNSS Logger does not seem to log this data in the file, but it shows it on the logging screen as hAcc=1 vAcc=2

This is what GNSS Logger uses in the file:
# Fix,Provider,LatitudeDegrees,LongitudeDegrees,AltitudeMeters,SpeedMps,AccuracyMeters,BearingDegrees,UnixTimeMillis,SpeedAccuracyMps,BearingAccuracyDegrees,elapsedRealtimeNanos

barbeau, 2021-10-07:
@tarquinwj Thanks for flagging this! So it looks like Google likely updated their logging code between the OSS version on GitHub and the current closed source version on Google Play that's available for download. GPSTest logging is derived from the OSS version on GitHub, and indeed it seems like that version is capped to 6 decimal places, although unintentionally.

Here's the logging code that transforms the location from raw values to String to be logged:

```
                String locationStream =
                        String.format(
                                Locale.US,
                                "Fix,%s,%f,%f,%f,%f,%f,%d",
                                location.getProvider(),
                                location.getLatitude(),
                                location.getLongitude(),
                                location.getAltitude(),
                                location.getSpeed(),
                                location.getAccuracy(),
                                location.getTime());
```

I assume Java `String.format()` method with `%f` rounds to 6 decimal places by default, although I can't find any documentation saying this.

And indeed, here are my logs showing the difference.

*GPSTest v3.9.16:*

```
Fix,gps,28.071728,-82.426516,126.004150,0.000000,21.345995,1633630855000
Fix,gps,28.071432,-82.426521,18.931580,0.000000,18.966507,1633630856000
Fix,gps,28.071340,-82.426583,1.942596,0.000000,17.772997,1633630857000
```

*GnssLogger v3.0.0.10*

```
Fix,GPS,28.0713376,-82.4265532,17.389465,0.000000,5.957377,0.000000,1633630386000,0.17492856,0.0,92784767961858
Fix,GPS,28.0713375,-82.4265533,17.407959,0.000000,5.668525,0.000000,1633630387000,0.25806975,0.0,92785767396493
Fix,GPS,28.0713374,-82.4265535,17.386108,0.000000,5.417323,0.000000,1633630388000,0.11313708,0.0,92786767240295
```

I'll work on fixing this so the full precision is logged, and do the same for other logged decimal values too.

closed 2021-11-18 by barbeau

tarquinwj, 2022-07-18:
Just checking, but is there a chance that this fix could have caused it to output scientific notation when the numbers are close to 0?
Eg. 0.00000000000000000000001 latitude, longitude and height?

barbeau, 2022-07-18:
@tarquinwj It's possible I missed something. Could you please open a new issue with the data you're seeing that has the scientific notation? If you can include the full log files that would be great, but if not even the field names and values where you're seeing this would help.