barbeau/gpstest #554: Save coordinates with a name

tarquinwj, 2021-10-04:
I would like to be able to save point coordinates in full precision, with a label/name, eg. to save points of interest (every other app that can do this insists on shaving off precision). This currently requires me to use "Share" in GPSTest, enable altitude, then pass the coordinates to another app where I can then save it with a name. It would be great to cut out the middle man.

Data that is needed:
Some Name,50.12340123456789,-3.12340123456789,101.12340123456789
Stored in a file, or just in a list in GPSTest would be fine.

Currently using GPSTest v3.9.14, Android 11, API Level 30

Possibly beyond scope, but this is a feature request.

wiltonp80, 2021-10-07:
+1 For this...

I use GPS Test to test certain locations at certain times and record the results to log files. I then use this to check against what local nearby base stations can see to match the two records. I have to ensure files are renamed to include the site details but it would be far easier to include the name in the log file directly from the app. 

Higher precision would also be welcomed. An extra decimal place or 2 would be welcomed; raw location precision would be even better.

tarquinwj, 2022-11-15:
Please re-open this one. GitHub won't let me.

ri984, 2022-11-15:
Test

closed 2023-12-26 by stale[bot]