barbeau/gpstest #565: Accuracy: Allow long-running background tests

barbeau, 2021-11-11:
**Is your feature request related to a problem? Please describe.**
After background service implementation in https://github.com/barbeau/gpstest/pull/550, currently Accuracy mode data is still contained in the Activity. So, if you put the Activity in the background it will eventually be killed, which destroys the test data even if the service and flows are still active.

**Describe the solution you'd like**
Allow long-running tests in the background when the screen isn't visible. This requires moving the accuracy data into the repo layer so it's retained while the service is running, even if the Activity is destroyed.

danieldjewell, 2023-05-27:
Random thought - Could the phantom process killer "feature" added in Android 12(?) be related here? (I know that it was kind of a huge problem for termux ....... https://github.com/termux/termux-app/issues/2366)

barbeau, 2023-06-06:
@danieldjewell Good thought, but this specific issue is a design upgrade that I havent done yet (moving the data outside the Activity lifecycle.

It's certainly possible that the phantom process killer could be an issue with GPSTest v3.10.2 running/logging in the background on some devices, but I havent seen any complaints yet. If anyone has a complaint, let's track that as a new issue.

barbeau, 2023-12-23:
Not stale

barbeau, 2026-03-04:
Not stale