barbeau/gpstest #566: Document "# sats" field

Ionic, 2021-11-13:
Currently, the "# sats" field in the status view shows a rather cryptic number of `X/Y` or `X/Y/Z`.

While X/Y might be easily explainable meaning `X` = satellites the device has a fix on and `Y` = total number of **visible** satellites (which differs from "total satellites" in that subtle manner), `X/Y/Z` is more difficult to grasp, and frankly, I haven't.

You can probably easily explain it here, but it should also be added to the FAQ/documentation.

I believe that the `X/Y/Z` scheme is used for multi-frequency GNSS.

Ionic, 2021-11-13:
Ah, and it turns out that I'm stupid. This is documented in the source code at https://github.com/barbeau/gpstest/blob/23b482d505628ae8f62633f46e34ade1c94a6d9f/GPSTest/src/main/res/values/strings.xml#L413 and even translated into multiple languages.

This said, though, where would this information show up? I've used the "Help" menu option, but this full help text newer showed up there. Rather, `main_help_options` is used, which only contains the `What's New` and `About` items.

So maybe this missing information is a bug because the help functionality isn't working correctly?

Ionic, 2021-11-13:
And I'm stupid again. The HelpActivity is correctly launched if selecting the "About" option and the resulting page actually shows the full help text, including device and application information.

I probably just missed the field documentation in the information flood, but it's there alright. Closing, since there's nothing to do here, really.

closed 2021-11-13 by Ionic

barbeau, 2021-11-15:
Haha, thanks. Yes, I agree this takes some explaining and is probably due to strange OEM handling of the data. On many devices I think the last two numbers will be the same, but on some they definitely aren't.